motioneye-project/motioneyeos #1296: Logitech C920 not working

bdaniel7, 2017-12-27:
I'm trying to run montioneye 20171008 on a RasPi 3 B, but there is no image.
motionEye Version | 0.37.1
Motion Version | 4.0.1+git2a140fa
OS Version | motionEyeOS 20171008

The camera is detected as /dev/v4l/by-id/usb-046d_HD_Pro_Webcam_C920_D9A1D1AF-video-index0 of type V4L2 Camera.
In the video camera placeholder it just says "Connection to camera lost since..".
I tried to play with various settings but without any luck.

In dmesg.log I see this:

[Thu Jan  1 02:00:02 1970] usb 1-1.5: New USB device found, idVendor=046d, idProduct=082d
[Thu Jan  1 02:00:02 1970] usb 1-1.5: New USB device strings: Mfr=0, Product=2, SerialNumber=1
[Thu Jan  1 02:00:02 1970] usb 1-1.5: Product: HD Pro Webcam C920
[Thu Jan  1 02:00:02 1970] usb 1-1.5: SerialNumber: D9A1D1AF
[Thu Jan  1 02:00:02 1970] uvcvideo: Found UVC 1.00 device HD Pro Webcam C920 (046d:082d)
[Thu Jan  1 02:00:02 1970] uvcvideo 1-1.5:1.0: Entity type for entity Processing 3 was not initialized!
[Thu Jan  1 02:00:02 1970] uvcvideo 1-1.5:1.0: Entity type for entity Extension 6 was not initialized!
[Thu Jan  1 02:00:02 1970] uvcvideo 1-1.5:1.0: Entity type for entity Extension 12 was not initialized!
[Thu Jan  1 02:00:02 1970] uvcvideo 1-1.5:1.0: Entity type for entity Camera 1 was not initialized!
[Thu Jan  1 02:00:02 1970] uvcvideo 1-1.5:1.0: Entity type for entity Extension 8 was not initialized!
[Thu Jan  1 02:00:02 1970] uvcvideo 1-1.5:1.0: Entity type for entity Extension 9 was not initialized!
[Thu Jan  1 02:00:02 1970] uvcvideo 1-1.5:1.0: Entity type for entity Extension 10 was not initialized!
[Thu Jan  1 02:00:02 1970] uvcvideo 1-1.5:1.0: Entity type for entity Extension 11 was not initialized!
[Thu Jan  1 02:00:02 1970] input: HD Pro Webcam C920 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/input/input1

Another Logitech webcam that I'm using, an older QuickCam is detected as 
/dev/v4l/by-id/usb-046d_08ce_6A727FC2-video-index0 and it's working.

I see that Linux, generally, properly detects the webcam.

Any ideas as to how to make it work?

ccrisan, 2017-12-28:
This looks like a driver or a hardware problem. I can't help you with that. Sorry.

closed 2017-12-28 by ccrisan

leetomblin, 2017-12-28:
Same issue with the C920 since updating this morning, so don't think it's a hardware problem.

motionEye Version | 0.37.1
Motion Version | 4.0.1+git2a140fa

ccrisan, 2017-12-28:
@leetomblin does it work with plain Raspbian?

leetomblin, 2017-12-28:
Haven't tried it for a while - but it was doing, until I moved over to motioneye.

chumstick, 2017-12-28:
I can confirm that on a Pine64, my C620 worked on Debian - but getting the same grey screen error through MEOS. Tried as recently as this morning.

bdaniel7, 2017-12-28:
Tested with 2017-11-29-raspbian-stretch, and the webcam is detected and can be used.
Linux version 4.9.59-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1047 SMP Sun Oct 29 12:19:23 GMT 2017

[    5.112188] usb 1-1.5: New USB device found, idVendor=046d, idProduct=082d
[    5.112203] usb 1-1.5: New USB device strings: Mfr=0, Product=2, SerialNumber=1
[    5.112211] usb 1-1.5: Product: HD Pro Webcam C920
[    5.112219] usb 1-1.5: SerialNumber: D9A1D1AF
[    5.166150] uart-pl011 3f201000.serial: no DMA platform data
[    5.227687] media: Linux media interface: v0.10
[    5.287070] Linux video capture interface: v2.00
[    5.494297] usbcore: registered new interface driver snd-usb-audio
[    5.495628] uvcvideo: Found UVC 1.00 device HD Pro Webcam C920 (046d:082d)
[    5.499398] uvcvideo 1-1.5:1.0: Entity type for entity Processing 3 was not initialized!
[    5.499418] uvcvideo 1-1.5:1.0: Entity type for entity Extension 6 was not initialized!
[    5.499429] uvcvideo 1-1.5:1.0: Entity type for entity Extension 12 was not initialized!
[    5.499440] uvcvideo 1-1.5:1.0: Entity type for entity Camera 1 was not initialized!
[    5.499451] uvcvideo 1-1.5:1.0: Entity type for entity Extension 8 was not initialized!
[    5.499462] uvcvideo 1-1.5:1.0: Entity type for entity Extension 9 was not initialized!
[    5.499473] uvcvideo 1-1.5:1.0: Entity type for entity Extension 10 was not initialized!
[    5.499484] uvcvideo 1-1.5:1.0: Entity type for entity Extension 11 was not initialized!
[    5.500934] input: HD Pro Webcam C920 as /devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0/input/input2
[    5.504531] usbcore: registered new interface driver uvcvideo
[    5.504543] USB Video Class driver (1.1.1)

ccrisan, 2017-12-28:
A new prerelease is coming out soon. It comes with updated kernel/drivers and firmware so it may fix the issue you're having.

bdaniel7, 2017-12-28:
Great, thank you!

leoheck, 2018-02-22:
Any news if this issue is already solved? I have the same problem.

chumstick, 2018-02-22:
I never resolved it. Wound up digging in some boxes for a different camera - I’d forgotten about this thread, honestly. Willing to help where I can, though! I just didn’t have the time that day and then forgot.