motioneye-project/motioneyeos #1297: SSH not working in latest version

Ikkeme, 2017-12-27:
The wiki tells me I can ssh into motioneyeos by using the regular port. Somehow I get "connection refused" whatever username I try. Also telnet fails with the same error.The interface says that SSH is on but I doubt that is true. It doesn't matter how the switch is set because SSH is always off.

ccrisan, 2017-12-28:
Please attach your boot.log file.

Ikkeme, 2018-01-29:
Please accept my apologies. I apparently completely forgot about opening this issue. The boot.log file says amongst other things: * Starting sshd: failed. All other things in the file are marked as 'done'.

ccrisan, 2018-01-30:
I recommend writing the OS image from scratch then. The data partition might have gotten corrupted.

Ikkeme, 2018-02-02:
Thanks, I can probably try that. Just out of curiosity: what needs to be in the data partition in order to get ssh working? I mean what file(s) and what content of the file(s)?

ccrisan, 2018-02-02:
If you do a `cat /etc/sshd_config`, you'll see the following contents:

    Port 22
    ListenAddress 0.0.0.0

    HostKey /data/etc/ssh_host_rsa_key
    HostKey /data/etc/ssh_host_dsa_key
    HostKey /data/etc/ssh_host_ecdsa_key
    HostKey /data/etc/ssh_host_ed25519_key

    PermitRootLogin yes
    PermitEmptyPasswords yes
    AuthorizedKeysFile /data/etc/ssh_authorized_keys
    UseDNS no
    Banner /var/cache/sshd_banner

    UsePrivilegeSeparation sandbox
    Subsystem sftp /usr/libexec/sftp-server
    IPQoS cs0 cs0

You'll get a gist of what needs to be present in `/data/etc`.

Ikkeme, 2018-02-02:
Elaborating on this: I noticed I can get an archive with the config files via the webinterface by using the configuration backup button. The hostkey files mentioned in your post above are present. There's no ssh_authorized_keys file and there's no ssh_config and sshd_config present. I suspect at least those last two should be present, right?

Ikkeme, 2018-02-04:
Where can I find the gist you mentioned above?

ccrisan, 2018-02-05:
>  there's no ssh_config and sshd_config present

Do you mean there's no `/etc/sshd_config` present on your system?

Ikkeme, 2018-02-06:
I meant there's no `/data/etc/sshd_config` present. I assumed it should be present on the data partition.

Ikkeme, 2018-02-06:
Right, I've decided to update to version 20180101 even though this isn't a stable version. But it seems to be better than the current stable version. SSH works now and as I expected there's now a `/data/etc/sshd_config` file present. There's only one difference. Above you mentioned the presence of the line `UsePrivilegeSeparation sandbox` in that file however in my file that one's not present.

Another still recurring problem is that wireless does not come up at first start. This was the problem in the previous version and still in the current version. I always have to pull the dongle and re-insert it. But probably I can better open a new issue for this problem.

jasaw, 2018-02-06:
@Ikkeme How did you move to version 20171008? Did you do firmware upgrade or fresh install into SD card?

ccrisan, 2018-02-08:
@Ikkeme `UsePrivilegeSeparation` is a deprecated SSH option and has been removed. You don't need to care about it.

> Another still recurring problem is that wireless does not come up at first start.

Is this fully reproducible? What does the boot.log file look like? Are you sure you don't have some hardware/powering issues there?

Ikkeme, 2018-02-09:
@jasaw I didn't move to version 20171008 but I moved to 20180101. In Expert Settings I've set Enable Prereleases to "on". Then I did a Software Update check and it came up with the 20180101 version.

Ikkeme, 2018-02-09:
@ccrisan It's not a matter of power issues because the wifi dongle is inserted in a powered hub so that must be sufficient. It might be however a hardware issue. boot.log says Starting wpa_supplicant: failed and after reinsertion is says Starting wpa_supplicant: done. So it did a restart. in dmesg.log I did find something interesting. It says:

`This Realtek USB WiFi dongle (0x0bda:0x818b) is untested!`
`Please report results to Jes.Sorensen@gmail.com`
`Vendor: Realtek`
`Product: 802.11n NIC`

Then it's dumping efuse (whatever that might be) to dmesg.log and after that it says:

`rtl8xxxu: Loading firmware rtlwifi/rtl8192eu_nic.bin`
`usbcore: registered new interface driver rtl8xxxu`
`

ccrisan, 2018-02-09:
@Ikkeme what board/model is this?

Ikkeme, 2018-02-11:
It's a Model B Revision 2.0 512 MB.

ccrisan, 2018-02-12:
Can you try and see if it works well with Raspbian? If it does, it may be a matter of kernel version/wifi firmware in motionEyeOS.

Ikkeme, 2018-02-12:
It did work in Raspbian although I don't remember if I did something special to make it work.

Related to that you might be interested in this thread on the raspberry pi forum:
`https://www.raspberrypi.org/forums/viewtopic.php?f=45&t=103989&sid=51fc99e1b07014f7c83200348498c71f&start=200`.
There's a driver available for this dongle but it's a bit of pain to use. Due to the way motioneyeos is setup it has to be installed over and over again until it is integrated in the underlying operating system.

ccrisan, 2018-02-16:
Unfortunately I'm not willing to maintain out-of-tree kernel drivers in motionEyeOS. The kernel source, version and configuration will be kept up to date with Raspbian.

closed 2018-02-16 by ccrisan