motioneye-project/motioneyeos #1298: Motion show frame changes showing "-"

drbeckma, 2017-12-28:
For my motioneyeos set up, I only see numbers from the 'show frame changes' option for about 2-3 seconds, they vary from 0-100 or upwards of 90k depending on settings, but then it will switch to that hyphen, only going back to numbers when I change settings and hit apply, and even then only for a few seconds before turning back into the hyphen. 

Any ideas on what's up with that?

Thanks!

[image]

SteveCohen, 2018-01-06:
The number in the top RHS indicates the number of pixels that have changed since the last frame = amount of motion. It'll show "-" when there's no motion detected. [official docs are here](https://motion-project.github.io/motion_config.html#text_changes).

ccrisan, 2018-01-07:
@drbeckma, @SteveCohen pointed you to the correct reference. If you don't want to see that, just disable the *Show Frame Changes* option.

closed 2018-01-07 by ccrisan