motioneye-project/motioneyeos #1303: help wifi on PI 3

bryands1, 2017-12-29:
Hello,
I was wondering if someone could tell me to disable the built in wifi so I get an USB wifi setup or how to configure 2 wifi souces in motioneyeos?

thanks

robotzero, 2017-12-30:
From the wiki:
https://github.com/ccrisan/motioneyeos/wiki/modprobe.conf

keyword blacklist will disable specified driver from autoloading.

As for using 2 wifi sources I am not sure. From my experience when I did connect external USB wifi dongle to my PI 3 it showed as the only wlan interface and my internal wifi somehow turned off automatically.

ccrisan, 2017-12-30:
@robotzero is right: the correct way of disabling the builtin wifi module (and any other module, for that matter) is the [modprobe.conf](https://github.com/ccrisan/motioneyeos/wiki/modprobe.conf) way: just add `blacklist brcmfmac` (the name of the builtin RPi wifi driver) to `/data/etc/modprobe.conf` and you should no longer see the default wifi interface.

motionEyeOS was designed to have one single network connection. It could work with more network connections, in theory, but init scripts will not to operate well like that. You're probably better off using Raspbian if you must have multiple network interfaces at the same time.

One more thing: there's an [os.conf](https://github.com/ccrisan/motioneyeos/wiki/os.conf#os_wlan) file that allows you to change the wlan interface using `os_wlan`. You can set it to `wlan1` to instruct it to use the secondary interface by default.

You can then add [custom commands at startup](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-custom-startup-commands) to manually configure your other interface(s).

closed 2017-12-30 by ccrisan

askdesigners, 2018-02-25:
Can someone provide some pointers on installing new drivers?? wget, apt-get are not installed, so that leaves you with curl which is not clever enough to follow redirects on links like Dropbox or github...

rsync from another machine results in "read-only file system error" 

Tips?

jasaw, 2018-02-25:
@askdesigners Raspbian with motionEye might be more suitable for you in this case. Another option is building motionEyeOS from source, https://github.com/ccrisan/motioneyeos/wiki/Building-From-Source

askdesigners, 2018-02-26:
Thanks! @jasaw