motioneye-project/motioneyeos #1304: Automatic Brightness

OldHolborn01, 2017-12-31:
Hello,

I suspect that "automatic brightness" does not work. With this setting, the image will pump from dark to light and back again, but it will not snap.

The problem I have both in Raspi 1, 2 or 3 in the various recent releases of motionEos

Can anybody confirm this?

Best regards
Reinhard

OldHolborn01, 2017-12-31:
I tried this with V4L2 Camera
...

causalityzach, 2018-01-01:
I suspect your camera already has autobrightness so the camera and Pi are caught in a loop of trying to adjust to each other. That is what happened in my case. Please see this page with a similar complaint I found when I googled the issue myself: https://github.com/ccrisan/motioneyeos/issues/864

closed 2018-06-05 by jasaw