motioneye-project/motioneyeos #1305: RasPi2 + NoIR Camera Module V2

smartsoldier, 2018-01-01:
The camera seems to be detected but the preview just shows the camera behind a non-smoking sign icon. vcgencmd get_camera shows 1 and 1. I have tried with fast network camera and not. No authentication. /etc/modules already has:

bcm2835-v4l2 max_video_width=1920 max_video_height=1200
bcm2835-wdt
i2c-bcm2708
i2c-dev

jasaw, 2018-01-14:
Were you running an older motionEyeOS version where the system was running fine, then upgrade to a later motionEyeOS version and camera stopped working after that? If yes, from which version to which version?
If no, have you tried a different SD card?

jasaw, 2018-06-05:
Closing issue due to lack of follow up.

closed 2018-06-05 by jasaw