motioneye-project/motioneyeos #1306: Occasional Hangs requiring power cycle

Bra1nK, 2018-01-01:
using RPI3 with Pi NoIR camera and motionEye Version | 0.37.1, Motion Version | 4.0.1+git37b3595,OS Version | motionEyeOS 20171003
Has happened 3 times in as many weeks, the relevant part of the messages log is reproduced below, it would seem the watchdog fails to do a restart then the camera won't reconnect.
Jan  1 03:02:45 motionEye3 daemon.info dhclient: DHCPREQUEST on eth0 to 192.168.1.100 port 67
Jan  1 03:02:45 motionEye3 daemon.info dhclient: DHCPACK from 192.168.1.100
Jan  1 03:02:45 motionEye3 daemon.info dhclient: bound to 192.168.1.12 -- renewal in 16599 seconds.
Jan  1 03:14:00 motionEye3 cron.info crond[942]: USER root pid 30531 cmd /usr/sbin/logrotate /etc/logrotate.conf
Jan  1 07:22:00 motionEye3 user.err motion: [0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, trying to do a graceful restart
Jan  1 07:23:00 motionEye3 user.err motion: [0:motion] [ERR] [ALL] main: Thread 1 - Watchdog timeout, did NOT restart graceful,killing it!
Jan  1 07:23:02 motionEye3 user.err motion: [0:motion] [ERR] [ALL] main: cleaning Thread 1
Jan  1 07:23:16 motionEye3 user.warn motion: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Jan  1 07:23:19 motionEye3 user.err motion: [1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format. VIDIOC_S_FMT: : Invalid argument
Jan  1 07:23:22 motionEye3 user.err motion: [1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format. VIDIOC_S_FMT: : Invalid argument
Jan  1 07:23:22 motionEye3 user.err motion: [1:ml1] [ERR] [VID] VIDIOC_TRY_FMT failed for format v4l2_set_pix_format: Invalid argument
Jan  1 07:23:22 motionEye3 user.err motion: [1:ml1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
Jan  1 07:23:32 motionEye3 user.err motion: [1:ml1] [ERR] [ALL] motion_init: Error capturing first image
Jan  1 07:23:32 motionEye3 user.err motion: [1:ml1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
Jan  1 07:23:40 motionEye3 user.warn motion: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jan  1 07:23:43 motionEye3 user.warn motion: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Jan  1 07:23:46 motionEye3 user.err motion: [1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format. VIDIOC_S_FMT: : Invalid argument
Jan  1 07:23:49 motionEye3 user.err motion: [1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format. VIDIOC_S_FMT: : Invalid argument
Jan  1 07:23:49 motionEye3 user.err motion: [1:ml1] [ERR] [VID] VIDIOC_TRY_FMT failed for format v4l2_set_pix_format: Invalid argument
Jan  1 07:23:49 motionEye3 user.err motion: [1:ml1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
Jan  1 07:23:49 motionEye3 user.err motion: [1:ml1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
Jan  1 07:23:50 motionEye3 user.warn motion: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
Jan  1 07:23:53 motionEye3 user.warn motion: [1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
Jan  1 07:23:56 motionEye3 user.err motion: [1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format. VIDIOC_S_FMT: : Invalid argument
Jan  1 07:23:59 motionEye3 user.err motion: [1:ml1] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format. VIDIOC_S_FMT: : Invalid argument
Jan  1 07:23:59 motionEye3 user.err motion: [1:ml1] [ERR] [VID] VIDIOC_TRY_FMT failed for format v4l2_set_pix_format: Invalid argument
Jan  1 07:23:59 motionEye3 user.err motion: [1:ml1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
Jan  1 07:23:59 motionEye3 user.err motion: [1:ml1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
Jan  1 07:24:00 motionEye3 user.warn motion: [1:ml1] [WRN] [ALL] motion_loop: Retrying until successful connection with camera

wgbiker, 2018-01-14:
Have the same problem on RPi3 with Camera Module v2.
MotionEye Version  0.38, 
Motion Version  4.1.1
motionEyeOS Version 20180101

Bra1nK, 2018-01-14:
I increased the GPU memory to 256 in 'Expert Settings' and it hasn't happened since, maybe coincidence but time will tell.

ccrisan, 2018-01-16:
Actually it's not a bad idea to increase the default GPU mem a bit, at least for 1GB models. I'll make 256 the default.

wgbiker, 2018-01-16:
Thanks for advice, I'm going test this configuration.

wgbiker, 2018-01-17:
The problem is still present, even after increasing GPU Memory to 256. 
It hanged just after I logged in to the admin panel. Other times it happened in similar circumstances.

Bra1nK, 2018-01-17:
Just noticed @wgbiker you're reporting motionEyyOS version 20180101, is that a misprint ?
I just checked for a software update and got 
"motionEye is up to date (current version: 20171008)"

wgbiker, 2018-01-17:
Yes, I do use pre-release version (20180101).

wgbiker, 2018-01-27:
It crashes on 20171008 as well. 
Frame Rate is set to 20fps, resolution 1280x720, motion detection enabled. 
Are these settings too demanding for RPi3?

Bra1nK, 2018-01-27:
I'm at 1024 x 768, 15fps with motion detection and it's been rock solid since I increased the GPU memory to 256

ccrisan, 2018-02-06:
https://github.com/ccrisan/motioneyeos/commit/63d558b995d1bc0bcb461d3ed97ecc7534b0386f increases default GPU mem to 256 on RPi 2 and 3.

closed 2018-02-06 by ccrisan

jasaw, 2018-02-06:
@Bra1nK @wgbiker If you run `vcgencmd get_mem malloc` and `vcgencmd get_mem reloc` commands, you'll be able to see how much free memory is available to the GPU.
- malloc shows free gpu memory in malloc heap
- reloc shows free gpu memory in relocatable heap

Keep polling it and make sure it doesn't go near zero ever. This may help you optimize GPU memory allocation.