motioneye-project/motioneyeos #1312: No WiFi connection on RPi 0 W

zen, 2018-01-03:
Hi,
I followed steps in the instruction to install manually using Etcher with no luck. Then I used suggested script for writing image providing WLAN details, still no luck. RPI just blinks LED twice and nothing then. Checked with several different cards
Using Raspberry i Zero W
Image: motioneyeos-raspberrypi3-20171008.img

I then flashed other images (OctoPrint) and they both work.

Unfortunately I don't have HDMI adapter to check console, I will get one

liijoh, 2018-01-03:
Your using the wrong image for pi zero. You need to use the image for Raspberry pi (A, B, A+, B+, compute module, zero, zero W)

fordorpi, 2018-01-14:
I am also trying to get a pi zero w set up and I used the motioneyeos-raspberrypi-20180101 image.  I added the file wpa_supplicant.conf with my proper wifi info.  
On the initial boot the system fails at "configuring wired network" and reboots.  
I have tried multiple SD cards with the same result.

jasaw, 2018-01-14:
Please follow the steps in the below wiki pages and report back in detail exactly what you did and what you observed.
https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration
https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems

P00PD0G, 2018-02-12:
I have the same issue with the RPi 0W. I see no change between when I tried without the file to when I added the file. It still tried to connect to a wired network and failed with no indication of attempting to connect to a wireless network.

fordorpi, 2018-02-12:
I had to change the file a little to get it to work.  I think the original has double quotes and I used single.  Here is what I used...with obvious changes to protect the innocent: 

country=us
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network={
 scan_ssid=1
 SSID='thisisyourSSID'
 PSK='thisisyourpassword'

jasaw, 2018-02-12:
@P00PD0G Wifi preconfiguration only happens **once at first boot**. If you have booted up the device without the wpa_supplicant.conf file, you'll have to reflash the SD card. Read the wiki links I posted in this issue.

P00PD0G, 2018-02-12:
I did but I missed that part, sorry. It’s working now. Thank you!

closed 2018-02-12 by jasaw