motioneye-project/motioneyeos #1313: Hardware acceleration / Low framerate

wusa88, 2018-01-04:
Hi,

20171008 is the newest version of MotioneyeOS. Is there the hardware acceleration included? 

ccrisan commented on 7 Jul 2017
https://github.com/ccrisan/motioneyeos/issues/1021#issuecomment-313600285

Or do i have to change to the version of jasaw?
https://github.com/jasaw/motioneyeos/releases

Thanks!

ccrisan, 2018-01-04:
20171008 does contain all of @jasaw's work on hardware acceleration and so does 20180101, if you care to test the latest prerelease :)

wusa88, 2018-01-04:
Thanks!

Where to find the last pre release ?

wusa88, 2018-01-04:
Got it: 
https://github.com/ccrisan/motioneyeos/releases

jasaw, 2018-01-14:
@wusa88 Did you manage to install the latest pre-release? Happy for me to close this issue?

farleydwnsu, 2018-01-16:
I have 2 USB cams setup, with one on a Raspberry Pi 3 and the other on a Raspberry Pi Zero w.  They are both on Wi-Fi, and both configured with the same settings through the web interface.  They are both set to 1280x720 30 fps in the video device settings and 10 fps in the movie settings.  They are both on their respective 20180101 image.  I have enabled prereleases in the expert settings.  They are both using h.264/OMX encoding.  I have also turned off all the services in the services section.  The Pi3 works great and is showing around 10/10 fps in the overlay.  The pi zero is showing around 4/1 fps in the overlay.

Is this to be expected when using a pi zero w?  That fps seems low to me.  I thought with the hardware acceleration being added that would go up.  If I drop the resolution on the pi zero it gets the fps up to about 7/6.

jasaw, 2018-02-28:
Closing issue as original question has been answered.

closed 2018-02-28 by jasaw