motioneye-project/motioneyeos #1315: save images

regolo76, 2018-01-04:
Hello, 
I installed motioneye for raspbian from github. Everything works except for saving images. I tested both on the microsd in /home/pi/image and on the USB stick.
I mounted sda1 and created a symbolic link. I configured the IP Cam:
**Storage device**: /home/pi/images/Camera1
**Root directory**: /

But it does not save images. Do you have any solution?
Thank you

regolo76, 2018-01-04:
**motioneye.log**
```
Jan 04 17:05:30 raspberrypi motion[5275]: [27399704:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Jan 04 17:05:30 raspberrypi motion[5275]: [2:ml2] [WRN] [NET] netcam_next: called with no data in buffer
Jan 04 17:05:30 raspberrypi motion[5275]: [1:ml1] [WRN] [NET] netcam_next: called with no data in buffer
Jan 04 17:05:30 raspberrypi motion[5275]: [3:ml3] [WRN] [NET] netcam_next: called with no data in buffer
Jan 04 17:05:31 raspberrypi meyectl[389]:  WARNING: mjpg client connection for camera 2 on port 8082 is closed
Jan 04 17:05:31 raspberrypi meyectl[389]:  WARNING: mjpg client connection for camera 3 on port 8083 is closed
Jan 04 17:05:31 raspberrypi meyectl[389]:  WARNING: mjpg client connection for camera 1 on port 8081 is closed
Jan 04 17:05:31 raspberrypi meyectl[389]:  WARNING: mjpg client connection for camera 4 on port 8084 is closed
Jan 04 17:05:32 raspberrypi motion[5275]: [3:ml3] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
Jan 04 17:05:33 raspberrypi meyectl[389]:  WARNING: Connect error on fd 22: ECONNREFUSED

Jan 04 17:05:30 raspberrypi motion[5275]: [27399704:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
Jan 04 17:05:30 raspberrypi motion[5275]: [2:ml2] [WRN] [NET] netcam_next: called with no data in buffer
Jan 04 17:05:30 raspberrypi motion[5275]: [1:ml1] [WRN] [NET] netcam_next: called with no data in buffer
Jan 04 17:05:30 raspberrypi motion[5275]: [3:ml3] [WRN] [NET] netcam_next: called with no data in buffer
Jan 04 17:05:31 raspberrypi meyectl[389]:  WARNING: mjpg client connection for camera 2 on port 8082 is closed
Jan 04 17:05:31 raspberrypi meyectl[389]:  WARNING: mjpg client connection for camera 3 on port 8083 is closed
Jan 04 17:05:31 raspberrypi meyectl[389]:  WARNING: mjpg client connection for camera 1 on port 8081 is closed
Jan 04 17:05:31 raspberrypi meyectl[389]:  WARNING: mjpg client connection for camera 4 on port 8084 is closed
Jan 04 17:05:32 raspberrypi motion[5275]: [3:ml3] [WRN] [NET] netcam_init_jpeg: no new pic, no signal rcvd
Jan 04 17:05:33 raspberrypi meyectl[389]:  WARNING: Connect error on fd 22: ECONNREFUSED
```

ccrisan, 2018-01-05:
Have you configured it to create still images? If yes, what does your configuration look like?

P.S. wrong project (you were probably looking for [motionEye](https://github.com/ccrisan/motioneye)).

regolo76, 2018-01-05:
Yes, I'm sorry to have written about the wrong project :-(.
I will write on motioneye :-)

Thank you