motioneye-project/motioneyeos #1316: Cannot get motion eye to boot from wpa supplicant

moiJee, 2018-01-05:
Hello, whenever i boot motion eye using this file, motion eye just reboots at the beginning due to no Ethernet port(using raspberry pi zero w). i cannot get into motion pie properly, as i am only able to use a mouse with raspberry pi (no keyboard). i later plan to remotely get into it using my laptop, but currently am concerned with just getting it going.
[wpa_supplicant.conf.txt](https://github.com/ccrisan/motioneyeos/files/1606872/wpa_supplicant.conf.txt)
i'm very new to all of this so i'm not too sure what to do.
thanks

sashsc, 2018-01-12:
Where do i put the `wpa_supplicant.conf` ?? in which directory ?

After writing the latest image to sd card, it looks like a complete system.

[image]

the screenshot is from windows explorer, directly after writing the image.

Where i have to put the wpa.conf file ?

thanx
Sash

jasaw, 2018-01-13:
@sashsc  Follow the instructions here:
https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration

Your screenshot shows the root filesystem partition. You need the "Boot" partition.

ozzeruk, 2018-01-14:
Also remember to save as a .conf file - not a .conf.txt file. In Notepad you might need to first click file type 'all files', to stop it from trying to add a .txt extension.

sashsc, 2018-01-14:
Dake for your answers. Of course, I have stuck to it. Have some experience with Linux, through FHEM.

Since nothing worked, I followed the instructions to manually install motioneye. This guide is from the WIKI.

As a basis I have taken the current RASPIAN STRECTH 20171129. Then I  installed the motioneye manually.

Have the RASPI Zero W then restarted. Have synonymous over the desktop, the RASPI ZERO W successfully brought to WLAN. The camera module is also switched on. The power adapter has 2.5A.

But over the IP of the Raspi Zero w nothing is attainable in the LAN.

Do you have an idea why, or which port needs to be used?

Thank you
Sascha

fordorpi, 2018-01-14:
I am having issues with the initial boot after loading the image and adding the wpa_supplicant.conf file.  When I boot the setup fails at configuring wired network.  Any suggestions?
I am trying again (fourth or fifth time).
Thanks

jasaw, 2018-01-14:
@sashsc Try probing port 80.

@fordorpi The wpa_supplicant.conf file is only read once at first boot. If you're still having wireless problem, wiki below might help.
https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems

sashsc, 2018-01-14:
Port 80 it didn't work. A portscan shows that only Port 22 is reachable

fordorpi, 2018-01-14:
I was able to get it to work.  This time I created a new wpa_supplicant file rather then using the one I use with FPP.  The only difference is when I followed the example on https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration
it has me using double quotes " rather than single ' that works for FPP
Great forum. Thanks for the help

jasaw, 2018-01-14:
@sashsc SSH into your Pi, see whether motionEye is running. Attach your motioneye logs here.

@fordorpi Glad to know your wifi works now. :-)

sashsc, 2018-01-14:
where can i found the logs ?

sashsc, 2018-01-14:
I have take a look at the motioneye.conf file

There was a port for tcp on 8765
Therefore i have reached the webif ip:8765

jasaw, 2018-01-14:
@sashsc cool, problem solved?

sashsc, 2018-01-18:
Yes

jasaw, 2018-01-18:
@ccrisan Any reason why you reopened this issue?

ccrisan, 2018-01-19:
By mistake maybe :)

closed 2018-01-19 by ccrisan