motioneye-project/motioneyeos #1317: 20180101 seems to wipe admin password

smithatlanta, 2018-01-06:
On a raspberry pi zero w using the motioneyeos-raspberrypi-20180101.img image, I initially logged in as admin with the blank password.  I then updated the admin password to a new password and then clicked apply.  I then attempted to log in with my new password and it fails.   I also tried the blank password again and that did not work either.

ccrisan, 2018-01-07:
Make sure you have cleared your browser's cache. This sounds a lot like #1116. After you have cleared your browser's cache and reinstalled the OS, if the problem persists, please get back to us with further details (step by step instructions on how to reproduce the problem, log files) and we'll look into it.

rtkatch, 2018-01-13:
These problems definitely still exist. I've had a couple of the different instances:

- I tried changing the user and admin passwords through the motioneyeos web interface, and I can no longer log in. Not even with '*****', and not even with a different browser or cache clearing.
- After a fresh install, I decided to not touch passwords at all, and after a couple of days I could no longer log in, through the web interface or through ssh. Not with admin or root. I didn't touch anything.

Perhaps you could point to an area in the software to investigate?

ccrisan, 2018-01-13:
It would be useful to recover some files from a broken system. This could be done using a PC with Linux, investigating logs and particularly the file `/etc/motion.conf` from the data partition, which contains the admin password hash.

Ultimately, if the problem is reproducible, you could recover the entire disk image and send it to me so I can identify the issue.

nicholastripp, 2018-03-04:
I've got the same issue on 20180224.

New install, logged in with admin and set a password. Rebooted.

After the first reboot I disabled FTP and SMB. Rebooted a second time after this change.

After the second reboot it won't accept the admin password that I'd previously set.

ccrisan, 2018-03-04:
@nicholastripp did you have a previous version of motionEyeOS running on that board (at that IP address)? I'm not talking about a firmware upgrade but rather a previous, completely separate setup that ran on that IP address. If that's the case, which version was it?

nicholastripp, 2018-03-05:
@ccrisan I did not have a pervious version installed. This was on my first device running motionEyeOS. It was a new build on a Raspberry Pi Zero W.

ccrisan, 2018-03-05:
@nicholastripp I see. So browser caching is probably not the cause here. Does it happen with any password (as simple as e.g. `12345`)? Can you please help us reproduce the issue, step by step?

nicholastripp, 2018-03-05:
I'll do my best.

1. Downloaded and copied 20180224 to an SD card via dd.
2. Created the wpa_supplicant.conf file and added it to the root of the filesystem on the card.
3. Put card in RPi Zero W and booted. Boot was successful.
4. Logged in via the web interface with user "admin" and no password.
5. Set an admin password of "k(9gKLXZ>DZDLEA[nJv7maAMGhDUc?Q"
6. Clicked "Apply" and waited for system restart.
7. Reloaded the web interface and authenticated with the credentials from step 5.
8. Opened settings and disabled FTP and SMB.
9. Clicked "Apply" and waited for system restart.
10. Reloaded the web interface and attempted authentication with the credentials from step 5. This is where I received the invalid user/password error.

I initially suspected the complexity of the password I used could be part of the problem. So I repeated the same steps above but this time in step 5 I set a password of "xUobdmVPjqTXX2dZQ9". This gave the same result. After the second restart the password was not accepted for admin.

After the second time I did authenticate to the web interface with the username "User" and no password, so the authentication mechanism doesn't appear to be broken.

mikefoley, 2018-03-07:
I'm seeing the same thing.. Glad it wasn't just me.. 

Build 20180224

ccrisan, 2018-03-08:
@nicholastripp I haven't been able to reproduce the issue, even though I tried many variants of what you described.

However, given your instructions, step 6 wouldn't ask for a reboot. Also, at some point you must have enabled Expert Settings. Can you please make a screencast of what you are doing? There might be something we're missing here. Also it may not be of importance, but:

 1. What does your wpa_supplicant.conf look like? Does it connect flawlessly at first boot?
 2. Can you reproduce this every time you're following the steps? Or is it something that occurs randomly?

@mikefoley the same questions apply, more or less, to you as well :)

mikefoley, 2018-03-10:
Supplicant is simple
     country=US
        update_config=1
        ctrl_interface=/var/run/wpa_supplicant
        
        network={
            scan_ssid=1
            ssid="SSID"
            psk="password"
        }

No issues connecting to WiFi.

Send me email and I'll send you a link to the screencast I took. The issue happens when you enter advanced settings and reboot.

mike@yelof.com

nicholastripp, 2018-03-17:
@ccrisan You're correct. I did enable expert settings at some point in that process. I'll run through it again and revise.

My supplicant file is the same as @mikefoley's. Seems to work fine.

If you want a second screencast I'm happy to provide one.

nicholastripp, 2018-03-17:
@ccrisan 

I just repeated this by doing (instead of from memory) and here are the steps to reproduce:

1. Loaded local site
2. Logged in with "admin" and no password
3. Selected menu icon in upper left
4. Set admin password of "k(9gKLXZ>DZDLEA[nJv7maAMGhDUc?Q" and clicked "Apply"
5. Logged in again with "admin" and password set in step 4
6. Selected menu icon in upper left
7. Enabled Advanced Settings
8. Set timezone to "America/New York"
9. Set hostname to "test"
10. Clicked "Apply"
11. Received prompt to reboot the system and clicked "Yes"
12. After reboot attempt to log in with "admin" and the password from step 4.
13. Receive "Invalid Credentials" error.

ccrisan, 2018-03-17:
@nicholastripp here are some ideas:

* can you please try with `*****` (five asterisks) as password and see if it actually works?
* can you try logging in via SSH and see if any of the imaginable passwords work?
* as a last resort, can you try recovering the `/etc/motion.conf` file from your data partition by using a Linux PC to read the SD card?

@mikefoley if you manage to reproduce the problem again with your setup, can you also try the above?

cquick97, 2018-04-03:
I was having the same issue, where it seemed like my admin password was getting wiped periodically/on boot, and stumbled upon this thread.

This is on a brand new installation of 20180401 on a Raspberry Pi 3 Model B. I basically installed using `writeimage.sh`, configuring Wi-Fi and a static IP. The board boots fine, and I can easily connect to it via Wi-Fi.

Then, I went to the web application page, and logged in as `admin` with no password. Then I set the admin password, along with timezone and hostname. When hitting "apply" the board asks to reboot, and when it comes back up I am able to log back in with the correct password. However, once another setting is changed, for instance changing static IP, and the board boots up fine, and I can connect on the new IP address. However, the password is now invalid!

I tried the above ideas, the password does seem to be set as `*****`. I can login using five asterisks in both the web interface, as well as SSH. When applying the configuration changes, it seems to include the placeholder text as the "new" password, thus overwriting with asterisks.

Hope that helped to figure out what is causing the issue!

cquick97, 2018-04-04:
I did a bit of digging. I'm not really sure how all the pieces fit together exactly, but the problem seems to start around here? 

https://github.com/ccrisan/motioneye/blob/fa8d9560b1c1c12279816a2d95fc322b6c128a73/motioneye/config.py#L701

    def main_dict_to_ui(data):
    ...
     # don't transmit password (or its hash) to the client;
     # instead transmit an indication of password being set
     if data['@admin_password']:
         ui['admin_password'] = '*****'
 
     else:
         ui['admin_password'] = ''
 
     if data['@normal_password']:
         ui['normal_password'] = '*****'
 
     else:
         ui['normal_password'] = ''

and here

https://github.com/ccrisan/motioneye/blob/fa8d9560b1c1c12279816a2d95fc322b6c128a73/motioneye/config.py#L670

    def main_ui_to_dict(ui):
    ...
    if ui.get('admin_password') is not None:
        if ui['admin_password']:
            data['@admin_password'] = hashlib.sha1(ui['admin_password']).hexdigest()

Like I said I don't totally understand all the pieces, but could this be causing a situation like

1. Default `data['@admin_password']` = `None`
2. User enters `ui['@admin_password']` = `Password123`, then hits save, triggering `main_ui_to_dict()`. This hashes and saves the correct password, saving from `ui['@admin_password'] -> data['@admin_password']`
3. At some point, the user does something to refresh the configuration, triggering `main_dict_to_ui()`. Probably by enabling advanced settings as the UI (probably?) needs to refresh in order to display the hidden advanced stuff, or rebooting. Since `data['@admin_password'] == Password123`, `if data['@admin_password']:` is `True`, so `ui['@admin_password']` _automatically gets set to_ `*****` when the page is loaded.
4. At this point, the user will probably want to make other configuration changes, and thus will need to save their configurations. This will again trigger `main_ui_to_dict()`, saving all the new and modified configurations. **However,** `ui['@admin_password']` **is now set to** `*****` **from when the configurations were initially loaded.**
5. This saves `*****` to `data['@admin_password']`, even though the user didn't change the field at all. It automatically got populated. If the user decided to edit the password field, it would also save the correct password to `data[]` (Likewise, a temporary workaround is to fill in the admin password field along with any other config changes, which seems to work for me in the small amount of time I've been trying).

This seems to be the case in my reading of the code, but I'm not really sure what to change to fix it. Perhaps a simple HTML password placeholder instead?

https://github.com/ccrisan/motioneye/blob/fa8d9560b1c1c12279816a2d95fc322b6c128a73/motioneye/static/js/main.js#L3398

    '<td class="dialog-item-value"><input type="password" name="password" class="styled" id="passwordEntry"></td>' +

would be

     '<td class="dialog-item-value"><input type="password" name="password" class="styled" id="passwordEntry" placeholder="*****"></td>' +

Then you wouldn't need to send `data['@admin_password'] -> ui['@admin_password']`, or `*****`, or anything at all, because it will already have a placeholder by the nature of the HTML element.

Damn, I didn't really intend to dive so deep here haha. I'm not exactly sure how to implement a fix, but hopefully this is enough to start with :)

ccrisan, 2018-04-04:
@cquick97 thanks for digging into this stuff so deep. It does seem like a very probable cause for what's been going on for some time with admin passwords. I'll take a deeper look myself over this entire piece of code and try to come up with a fix. Placeholders are indeed a great idea but that would just clear the password instead of setting it to five stars.

dax2112, 2018-05-30:
Yes I did the exact same steps as outlined above, sometimes 1 change or 2 and then no login available.  I will try the single IR light to see if that works.
Can I install MotionEye within NOOBS?

ccrisan, 2018-06-02:
https://github.com/ccrisan/motioneye/commit/818cd284ff7d6139618cf7dd21237c1417da3790 should fix this problem once and for always.

closed 2018-06-02 by ccrisan

jking1224, 2018-06-08:
This is still not fixed.
Using the release: 20180602
I I set up email notifications, the admin password is changed from (blank) to something else unknown.
The file: /data/etc/motion.conf then has a line like this: 
# @admin_password 2e4e0a9bf686e64e900aa8dd063729213656009d
instead of:
# @admin_password

ccrisan, 2018-06-09:
@jking1224 are you sure your browser has the latest `main.js` script? Normally it has a couple of protections that prevent any change to the password unless you explicitly type in stuff in that box.

jking1224, 2018-06-09:
I am not a programmer, I cannot help investigate this problem at a technical level such as knowing if I have the latest main.js.

Bottom line is that there IS a problem that when the user changes the settings for sending email notifications, something also occurs that the admin password is changed. In my case, it changed to something NOT five asterisks because I tried that. The admin password was not the password I set into the email notifications settings. So, I do not know what the new admin password had become and I was locked out, had to re-format and start over.

This is a bad problem that when user uses the software as intended, simply filling in the options provided, they get totally locked out and have to reformat their system.

My current workaround is that now, whenever I change ANY admin settings, I also re-type the admin password into the users area as the last step before I click "apply". 

A suggestion is to have two password boxes, the user must type the new password twice and the two values must match before the new value gets saved. This would protect against the user mis-typing and also protect against any logic flaw in the programming that also un-intentionally changes the password. 
Or another option is that "changing the admin password" is completely separate from all the other system options. Separate window/screen/menu choice. So, that changing the admin password is separate function on a single function screen. Changing any other option is completely separate area.

Release:  20180602 does not fix this issue "once and for always"

ccrisan, 2018-06-09:
@jking1224, please try refreshing the page using Ctrl+F5 and see if the problem persists. This would normally ensure that you actually run the latest JavaScript code.

I totally agree that this is a very inconvenient issue and that's why I'm trying to fix it. However I was never able to reproduce it myself. This is why I'm relying on the help of others. This being said, if the problem is fully reproducible on your side, would you be willing to give me some kind of remote access so that I can see what's actually going on?

cquick97, 2018-06-15:
https://crackstation.net/ reports that "2e4e0a9bf686e64e900aa8dd063729213656009d" is the SHA-1 hash of "24december", if that helps narrow anything down.

If I have any free time tomorrow I'll try to load a pi with the latest version and see if I can reproduce the issue again.