motioneye-project/motioneyeos #1318: Error while updating from 20171003 to 20171008

KyleKatarn31, 2018-01-06:
Today i tried to update my Raspberry 3 from 20171003 to 20171008

Step 1 : backup my configuration on my local PC

Problem 1 : using the web interface, the "update in progress" stayed forever. Power off/on restarted on initial configuration. Conclusion : I had to manually download ISO and burn in onto my SD card

Problem 2 : Starting my fresh new 20171008 ... there was NO mean to restore configuration ! I had to install a 1st camera to finally get access to the "Restore configuration" button. This should be available even with NO camera installed !

Problem 3 : After installing my 1st camera and accessing the "restore config" button, i did the action... but got a "Failed to restore configuration". Then I was asked to login again... and discovered that configuration has been successfully reloaded !

Thank you for the great work in providing and maintaining motionEyeOS. Should I help to understand what happen... please let me know !

ccrisan, 2018-01-10:
1. Whenever something goes wrong, useful errors can be found in the log files (most notably motioneye.log). Given that you're the first user to report upgrade problems from 20171003 to 20171008, I'm inclined to believe that it was an isolated problem. I assume you no longer have those logs.

2. That's weird. The Backup/Restore buttons have nothing to do with any camera. Are you sure you have enabled the Advanced Settings?

3. The restoration could have failed at the end at some place. Again, motioneye.log can tell us more about what went wrong. Did you do the backup with 20171003 or was it an older one?

KyleKatarn31, 2018-01-14:
2 . There is NO Backupd/Restore, nor advance settings available in web browser until you add 1st camera (am i wrong ?)

3. backup made with 20171003 just before upgrading

1. here comes my log. No obvious clue...
2018-01-06 17:04:23: [motioneye]     INFO: hello! this is motionEye server 0.37.1
2018-01-06 17:04:24: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2018-01-06 17:04:24: [motioneye]     INFO: cleanup started
2018-01-06 17:04:24: [motioneye]     INFO: wsswitch started
2018-01-06 17:04:24: [motioneye]     INFO: tasks started
2018-01-06 17:04:24: [motioneye]     INFO: mjpg client garbage collector started
2018-01-06 17:04:24: [motioneye]     INFO: smb mounts started
2018-01-06 17:04:24: [motioneye]     INFO: server started
2018-01-06 17:04:28: [motioneye]  WARNING: 403 GET /login/?_=1515258269697&_username=admin&_signature=e6d405fa1a6f7b2c230cf6c6d9b34835ce5a597b (192.168.0.254) 2.96ms
2018-01-06 17:04:33: [motioneye]    ERROR: authentication failed for user admin
2018-01-06 17:04:33: [motioneye]  WARNING: 403 GET /login/?_=1515258274603&_username=admin&_login=true&_signature=dfe89bb4d4e01d93ed8b1eda7264f6ff0570fb97 (192.168.0.254) 6.07ms
2018-01-06 17:04:39: [motioneye]    ERROR: authentication failed for user admin
2018-01-06 17:04:39: [motioneye]  WARNING: 403 GET /login/?_=1515258280577&_username=admin&_login=true&_signature=b2095f8e34df271fce2064ed9c3515fa0eb6fd45 (192.168.0.254) 8.69ms
2018-01-06 17:05:03: [motioneye]  WARNING: 403 GET /login/?_=1515258304708&_username=_&_signature=1f753ce12faa9ff7d6f0cdbb119184d0eecc52d9 (192.168.0.254) 2.14ms
2018-01-06 17:05:17: [motioneye]  WARNING: 403 GET /login/?_=1515258318170&_username=_&_signature=7f881695c15b9b39aac54e48d3585d6f0783a057 (192.168.0.254) 2.13ms
2018-01-06 17:05:34: [motioneye]  WARNING: 403 GET /login/?_=1515258335357&_username=_&_signature=a7ebbc03eb991558d5d25521fe332867379b3162 (192.168.0.254) 2.23ms
2018-01-06 17:05:45: [motioneye]  WARNING: 403 GET /login/?_=1515258346084&_username=_&_signature=02e14de5771cc4c1f7636c8572d70569ec662dc9 (192.168.0.254) 4.03ms
2018-01-06 17:06:03: [motioneye]  WARNING: 403 GET /login/?_=1515258364756&_username=_&_signature=e31c3dce7b84f6e4f8f237eaff11f79dabbb1a4d (192.168.0.254) 2.12ms
2018-01-06 17:07:45: [motioneye]     INFO: adding new camera with id 1...
2018-01-06 17:07:45: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2018-01-06 17:08:06: [motioneye]     INFO: restoring config from backup file
2018-01-06 18:08:08: [motioneye]     INFO: rebooting
2018-01-06 18:08:08: [motioneye]     INFO: executing "/sbin/reboot"
2018-01-06 18:08:09: [motioneye]     INFO: interrupt signal received, shutting down...
2018-01-06 18:08:09: [motioneye]     INFO: server stopped
2018-01-06 18:08:09: [motioneye]     INFO: tasks stopped
2018-01-06 18:08:12: [motioneye]     INFO: motion stopped
2018-01-06 18:08:12: [motioneye]     INFO: smb mounts stopped
2018-01-06 18:08:12: [motioneye]     INFO: bye!
2018-01-06 18:08:42: [motioneye]     INFO: hello! this is motionEye server 0.37.1
2018-01-06 18:08:45: [motioneye]     INFO: cleanup started
2018-01-06 18:08:45: [motioneye]     INFO: wsswitch started
2018-01-06 18:08:45: [motioneye]     INFO: tasks started
2018-01-06 18:08:45: [motioneye]     INFO: mjpg client garbage collector started
2018-01-06 18:08:45: [motioneye]     INFO: smb mounts started
2018-01-06 18:08:45: [motioneye]     INFO: server started
2018-01-06 18:08:47: [motioneye]  WARNING: 403 GET /config/0/get/?_=1515258528571&_username=admin&_signature=01cc45c0280667b08f4b10f2c3040d98c2de0a61 (192.168.0.254) 6.70ms
2018-01-06 18:08:49: [motioneye]  WARNING: 403 GET /config/0/get/?_=1515258530602&_username=admin&_signature=70fbab889e8f86b137c9f313868917d010161815 (192.168.0.254) 3.76ms
2018-01-06 18:08:51: [motioneye]  WARNING: 403 GET /config/0/get/?_=1515258532618&_username=admin&_signature=f45252a7072f5686a0a9553e935d1e226599963b (192.168.0.254) 5.46ms
2018-01-06 18:08:53: [motioneye]  WARNING: 403 GET /config/0/get/?_=1515258534634&_username=admin&_signature=5e58294b8145b43c0f584870e13924354bd72f31 (192.168.0.254) 4.09ms
2018-01-06 18:08:55: [motioneye]  WARNING: 403 GET /config/0/get/?_=1515258536650&_username=admin&_signature=ae02c30515ef48e72c85dafa41a0215abb00d7c6 (192.168.0.254) 6.61ms
2018-01-06 18:08:57: [motioneye]  WARNING: 403 GET /config/0/get/?_=1515258538681&_username=admin&_signature=59926572e1e0ade56d8dbdc31ccdfe93a7555f81 (192.168.0.254) 3.47ms
2018-01-06 18:08:59: [motioneye]  WARNING: 403 GET /config/0/get/?_=1515258540697&_username=admin&_signature=d82e871c3b096616e69edded9f23d93ce5f52d57 (192.168.0.254) 3.26ms
2018-01-06 18:09:01: [motioneye]  WARNING: 403 GET /config/0/get/?_=1515258542712&_username=admin&_signature=6e77fa4879fde32949d7fa6d64e48a77c3147c9c (192.168.0.254) 6.76ms
2018-01-06 18:09:03: [motioneye]  WARNING: 403 GET /config/0/get/?_=1515258544745&_username=admin&_signature=67c20ec18ed72605848cf25d4ac31a18b6ebe46d (192.168.0.254) 3.42ms
2018-01-06 18:09:05: [motioneye]  WARNING: 403 GET /config/0/get/?_=1515258546760&_username=admin&_signature=aa203155b46cd26ffba5f38c365e59ca450c7a40 (192.168.0.254) 2.86ms
2018-01-06 18:09:07: [motioneye]  WARNING: 403 GET /config/0/get/?_=1515258548776&_username=admin&_signature=523e6f2321412c2d3b9eb8008d61dc75a4a234cc (192.168.0.254) 16.15ms
[... cut ...]
2018-01-14 11:11:15: [motioneye]    ERROR: failed to create movie preview for /home/ftp/sdcard//2018-01-14/11-10-25.mp4: Command '[u'ffmpeg', u'-i', u'/home/ftp/sdcard//2018-01-14/11-10-25.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'12', u'-y', u'/home/ftp/sdcard//2018-01-14/11-10-25.mp4.thumb']' returned non-zero exit status 69
2018-01-14 16:02:47: [motioneye]    ERROR: failed to create movie preview for /home/ftp/sdcard//2018-01-14/16-01-59.mp4: Command '[u'ffmpeg', u'-i', u'/home/ftp/sdcard//2018-01-14/16-01-59.mp4', u'-f', u'mjpeg', u'-vframes', u'1', u'-ss', u'12', u'-y', u'/home/ftp/sdcard//2018-01-14/16-01-59.mp4.thumb']' returned non-zero exit status 69