motioneye-project/motioneyeos #1322: motioneyeos sound support by integrating/merging audio recorder

safetyfactorman, 2018-01-11:
Let me first applaud the genius of this project.  There is an application called audio recorder, which is similar to motion, but for sound, not video.  If integrated into motioneye (some or all of the features), might be a revolutionary solution.  Why, because decoupling audio from video increases the utility of the solution.

ccrisan, 2018-01-13:
There are no plans to add audio support to motionEye anytime soon. However, you mentioned the existence of a program similar to `motion`, but dealing with audio. Can you provide a link to it?

safetyfactorman, 2018-01-13:
Hi,I think audio-recorder is marooned at launchpad.net:https://launchpad.net/~audio-recorder/+archive/ubuntu/ppa/+packageshttps://launchpad.net/~audio-recorder/+archive/ubuntu/ppa/+packages
audio-recorder is a weird situation. The original developer who is identified by osmoma??? appears to have stopped development on it.
So far as I can tell, audio recorder is like motion for sound.it does not install on arm cpus,  If you could get this running on pi, WOW!!!I had a bitch of a time getting this installed on linux mint17/18 the first time.   I used this article to begin with, but the instructions are wrong.https://www.unixmen.com/install-audio-recorder-1-4-2-ubuntu-linux-mint-elementary-os-pear-os/for sudo-apt get install to work, you need the right repository.  I forgot to write down what works.  Once you have the right repository, it should just install.  Maybe you can contact osmoma??? directly, and liase with that person.  You probably know how to resolve this much easier than I can.
I currently get the error ''This ppa does not support xenial''.  Maybe a Linux Mint 18 specific issue?If you get it running on pi, pls let me know.  If you solve the xenial issue for Mint18, please also let me know, and I will write it down for next time.thanks,Mark----- Original Message -----

safetyfactorman, 2018-01-13:
One other thing for future consideraton - OpenHab integration.

ccrisan, 2018-01-14:
I'm sorry but this looks like a dead project to me.

closed 2018-01-14 by ccrisan

Osirus126, 2018-05-23:
i know a way you can stream audio input from one device to another using pulseaudio, for example from the a mic on the pi to another computer on the network as a pulseaudio input.. it is possible through configuration of pulseaudio. i know how to do it by using the gui. but i dont know how to do it on motioneyeos end.. i would install "pulseaudio" and "pavucontrol" and "paprefs" on both systems and then open paprefs and select "Make discoverable Pulseaudio network sound devices available locally" under "Network Access" Tab. Then under "Network Server" Tab, you must select "enable network access to local sound devices"  This would all have to be done on the MotionEye Side of things and then on the computer that you would like to capture the sound from.. you can probably implement something along those lines.. let me know what you think.

asmhack, 2018-05-24:
It's a good idea to have video with sound. Or at least ability to choose only video or video+sound.

safetyfactorman, 2018-05-25:
There is a project called audio recorder that gives everything that is needed to put sound into motioneye.  I suggested it before, but no takers.
https://launchpad.net/~audio-recorder/+archive/ubuntu/ppathis runs on linux mint.  try it out.  if all that is required is to recompile on arm, then it may be to motioneye for sound, what motion is for video.

asmhack, 2018-05-26:
@safetyfactorman 
> it may be to motioneye for sound, what motion is for video.
Not sure if it's reasonable, to have "motioneye for sound". As sound assistant better to use something like https://mycroft.ai/get-mycroft/ I believe you can configure some triggers there.

I'm just asking about adding sound recording as well as video, so we'll have the video+sound as the result. It may be not useful for continuous movies (for outdoor cams), but for motion triggered indoor cams it may be a great feature.

@ccrisan I can create new feature request, cause it's a bit different than this one. Should I?

Rathna-K, 2019-04-10:
It would be great if motioneye can include audio communication as well. Could replace loads of doorbells/ Kuna lights etc. @ccrisan I understand you closed this but would you reconsider?

ccrisan, 2019-04-12:
@Rathna-K I still have no plans to add audio support to motionEye, due to other, more critical tasks that are still yet to be done and due to the fact that Motion itself doesn't support audio.

waldorffsmama, 2019-10-10:
Thanks @ccrisan for all your work on MotionEye. I am utterly impressed by it and it helped me a great deal to integrate my rtsp-streaming cameras in node-red. 

I do agree with the other posters here that integrating sound would be a great plus. In case you get a chance to implement even a two-way sound feature a lot of your happy users will be even happier.

Aspire1Inspire2, 2019-12-04:
Thanks @ccrisan for this awesome software. It makes using motion like a breeze.
I understand it would be rational for you to implement audio support by packaging it from "motion" output. In another word, I understand it would be resonable to ask the motion developers to add the audio feature first.

I am currently using VLC to stream sound when needed.
It would be nice, off course, if we can have sound stored track in the record file.
But there seems to be a strong tendancy in the motion developers' opnion that sound is not necessary for survalliance use. There is minimal chance that we can get audio support in the foreseeable future.

ortigosano, 2020-06-03:
I really like motioneye and I would love it if it could record sound with the video as well. I think as a surveillance software recording the audio could help a lot if something happens. ( A theft ) . I will try another alternative because the audio it's a must for me. 

Please consider this a feature request. I would be happy to donate to the cause

mikeyb5753, 2020-06-11:
another +1 for audio, I too would also donate.

safetyfactorman, 2020-06-11:
I think that the best approach is to take the audio-recorder code, and merge it into motion.  This might be relatively easy to do.  The code repository for audio-recorder is:

https://launchpad.net/~audio-recorder

and is under active development.  It may be better to communicate with the audio-recorder team and ask them to integrate with motion, or contact the motion team, and have them interface to audio-recorder.

safetyfactorman, 2020-06-11:
Once integration of motion and audio-recorder takes place, it then becomes a part of the motioneye requirements.

DeadEnded, 2020-06-24:
As a work around (for anyone who wants to try it) I have worked with a few others to make a script that can add audio to recordings. I have only tested it on my own setup (there area at least 3 others using it that I am aware of) but am happy to try and help others.

https://github.com/DeadEnded/MotionEyeAudio

Cheers!
DeadEnd

userosos, 2020-06-29:
I think that need add audio recorder from web cam pls! It is very need! I use an ONVIF camera it has microphone. Need audio triggered also for start record video file with audio. Some web-cam has microphone too. Add audio pls!

originalknibbler, 2020-10-26:
I would donate some money too if audio will be added. But thank you so far the for Motioneye os :-)

mdPlusPlus, 2020-12-14:
@DeadEnded Does this work with a "Raspiaudio MIC+" on a Pi 4?  
https://raspiaudio.com/produit/mic

DeadEnded, 2020-12-17:
IDK to be honest.
But a few others have had issues with MotionEyeOS as it doesn't work the same as MotionEye (different seutp).  So my guess is it will work, but might time some modifications.

locofocos, 2021-04-06:
> Motion itself doesn't support audio.

On https://github.com/Motion-Project/motion/issues/384, the motion team suggested
> For now you can start audio stream manually using ffmpeg and ffserver. I'm doing that with V4L device built-in mic and streaming audio as mpeg2:

Let's suppose the motion team could build streaming audio support. Maybe that means a different container format, I'm not sure. If they did though, what do you think would be a more feasible approach to supporting streaming audio?

* (the poor man's approach) Have motionEyeOS continue to consume the existing mjpeg (?) stream from motion, and open a separate connection to the `v4l.ffm` audio stream whenever you're viewing a fullscreen stream. Add some minimal audio controls to the existing fullscreen motionEyeOS page. 
  * Whichever single board computer has a USB webcam hooked up to it, this computer would run a command like `ffmpeg -f alsa -ac 1 -i hw:2,0 http://192.168.0.5:1000/v4l.ffm` to produce the stream.
* (the nice native approach) The motion team produces a stream that contains both video and audio. Then motionEyeOS consumes this new type of stream/container, and adds some minimal audio controls (volume/mute).

Maybe as a low-cost way of implementing the first bullet point, what if the person setting up motionEyeOS was entirely responsible for setting up the `v4l.ffm` audio stream? motionEyeOS could simply provide another input field under the camera setting named "Audio Stream URL", where the admin would type in `http://192.168.0.5:1000/v4l.ffm` (for example). If that field is set, then when viewing a fullstream video stream page, motionEyeOS does a minimal amount of work to connect to the stream and play it while showing the video stream. Do you think that could work?

starbasessd, 2021-04-06:
@locofocos Please note that both the motion-project ticket (closed 4 years ago) and this one were closed, and this one was labeled 'won't fix', 3 years ago.