motioneye-project/motioneyeos #1323: Error while updating from 20171003 to 20171008

swb1990, 2018-01-13:
Hi,

I know there is a similar bug raised earlier but the dude simply did a fresh installation.
I have the corresponding log files and it seems to be programming error that caused the update to fail.
This is from my motioneye.log when i attempted to update from 20171003 to 20171008

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/1628193/motioneye.log)

Any help would be appreciated
Thank you!

ccrisan, 2018-01-13:
`Exception: firmware download failed` - basically there was an error downloading the new firmware. This could be caused by network connectivity issues or by github/s3 temporary problems. In any case I recommend retrying.

swb1990, 2018-01-14:
Thanks for the response ccrisan
I made another attempt today and i still could not update
i attempted to do a network connectivity check of my motioneye via SSH. i was able to ping from motioneye to sites such as google so it would seem it is not by connection issue.
any other possibilities?

ccrisan, 2018-01-14:
You can do the firmware update via command line. See [this](https://github.com/ccrisan/thingos/wiki/Firmware-Updates#the-fwupdate-command) for the commands that you have to type. It's basically a series of `fwupdate <command>` and is what motionEye does in the background when you start a firmware update.

swb1990, 2018-01-14:
so basically the following?
fwupdate download 
fwupdate extract
fwupdate flashboot
fwupdate flashreboot
fwupdate upgrade
in this exact order?
Thanks for the advice...i'll give it a try

jasaw, 2018-01-14:
@swb1990 I think you just need to give this command: `fwupdate upgrade 20171008`

ccrisan, 2018-01-14:
@swb1990 yes, with the small observation that you need to supply the version to `fwupdate download`.
@jasaw in theory yes, but (1) we're trying to identify the problem here, hence we're running it step by step and (2) the `upgrade` command was introduced only recently and it may not be available on @swb1990's current OS.

swb1990, 2018-01-15:
it appears to be a disk space problem....so that probably answers the whole issue.
is there any way to check the current disk usage without physically removing the card? 
i can't remember the size.
also, if i want to switch to a bigger card, i suppose it's not a simple case of copy-paste from the smaller card?

swb1990, 2018-01-15:
Filesystem           Type            Size      Used Available Use% Mounted on
/dev/root            ext4          174.3M    167.2M         0 100% /
devtmpfs             devtmpfs      433.4M         0    433.4M   0% /dev
tmpfs                tmpfs         437.9M     16.0K    437.9M   0% /dev/shm
tmpfs                tmpfs         437.9M    732.0K    437.2M   0% /tmp
/dev/mmcblk0p1       vfat           19.9M      8.5M     11.5M  42% /boot
/dev/mmcblk0p3       ext4          254.9M      7.3M    230.1M   3% /data
/dev/mmcblk0p3       ext4          254.9M      7.3M    230.1M   3% /home/ftp/sdcard
/dev/mmcblk0p3       ext4          254.9M      7.3M    230.1M   3% /home/ftp/storage

this is what i get after i run du -h

jasaw, 2018-01-15:
You can "copy" to the bigger card using the `dd` command. Put your current SD card into your computer, run dd to copy disk to hard drive, then put your new and bigger SD card into the computer, run dd to copy the image to the bigger SD card, then run gparted to expand the partition on the bigger SD card. Done.

swb1990, 2018-01-16:
thanks jasaw!
i'll give that a try

jasaw, 2018-01-16:
@swb1990 Expand the last partition, the right most partition in gparted.

swb1990, 2018-01-18:
will do! thank you!