motioneye-project/motioneyeos #1325: motionEye with ip6

glanghei, 2018-01-15:
Hi, everyone,
I want to run motionEye with ip6. But I can't get the configuration right. motionEye is not started with interfaces in /etc/network but with a script. Actually it's easy, you have to switch on the ip6 protocol with the entry "iface[wlan|eth]0 inet6 auto" in the file "interfaces". If someone has an idea in which script and in which line the entry should be made. 
I use Raspberry Zero W with Raspberry-Camera,  motionEye with the default configuration.

ccrisan, 2018-01-16:
IPv6 is disabled in motionEyeOS due to many issues raising from users with misconfigured networks. If you must use IPv6, I recommend installing Raspbian + motionEye.

closed 2018-01-16 by ccrisan

jordipalet, 2021-01-27:
Running latest motioneye version in a dual-stack box. If I set in motioneye.conf
listen to ::
it works with IPv6-only
tried ::, 0.0.0.0 and :: 0.0.0.0 and ::;0.0.0.0 and also those with "" and '', no luck ...

but doesn't work
What is the right config to make it dual stack?

I see that changing in /usr/local/python2.7/dist-packages/settings.py LISTEN = "" it works, but I guess is not the best place to configure it, as it may be modified by an update?

starbasessd, 2021-01-27:
Probably safe at this point, as they are actively working to get rid of pythin2.7 anyway.