motioneye-project/motioneyeos #1326: Enable / disable motion detection over web API

AxelRHD, 2018-01-16:
Is it possible to turn the motion detection on and off over http(s) requests? If not, it would be a nice feature to interact with home automation systems.

ccrisan, 2018-01-19:
Not yet.

doubledrat, 2018-08-08:
I would like this too.  When I'm at home, I don't want 100's of emails.  My phone uses tasker to send a command to disable emailing on my current system.  Is this possible yet with motioneyeos?

tjet-lw, 2018-08-22:
Same here, this would be very useful. Great application otherwise!!

AxelRHD, 2018-11-23:
MQTT would also be a nice solution for config changes.

kabadisha, 2019-01-11:
+1 for enabling API 👍

GhostOverload, 2019-02-08:
+1 for enabling API

kabadisha, 2019-05-21:
For those who find this thread. There is actually a way:
https://github.com/ccrisan/motioneye/issues/800