motioneye-project/motioneyeos #1330: rtsp: Unable to open video device

Marthinius, 2018-01-18:
Hello,

First I tried installing motioneye on ubuntu 16.04 with armbian without success (some error). 
So I installed a motioneye OS to OrangePi One. Motioneye has started successfully, I added ip camera over rtsp protocol. The motionEyeOS config detect my ip camera but it says unable to open device with gray window. I can open the stream with VLC correctly with that rtsp URL. 
So URL works with VLC but does not with motionEye. 

I tried not only ip camera, but also usb camera still the same problem: Unable to open video device.

Marthinius, 2018-01-18:
I reinstalled everything again ....Solved

frobgm, 2018-02-20:
Marthinius, how do you add rtsp protocol to motioneye os?
thanks yours help!

closed 2018-02-20 by ccrisan