motioneye-project/motioneyeos #1332: Recomended Power supplies

StuartOswald, 2018-01-18:
What are the recommended minimum power supply specs for running RPi2 and Zero W respectively?

jasaw, 2018-01-18:
It depends on what you connect to your Raspberry Pi and how fast a transient response you need.
Without any information on your hardware set up, I recommend using the official 2.5A Raspberry Pi power supply if you are powering from the micro USB connector. RS stock number: 909-8126.

PegFaceMan, 2022-06-15:
Hi, adding here rather than create new.
Using MotioneyeOS on Pi Zeros and a 3A+.  Looking to minimise power consumption for Solar, Pi-Juice and Powerbank.
HDMI off
LED off
Should be able to do this on MotionEyeOS right?

starbasessd, 2022-06-15:
LEDs:
https://github.com/motioneye-project/motioneyeos/wiki/Turn-Off-LEDs
Can't turn off the GPU (especially on a Pi) as it uses the GPU to process the video. If you don't connect a monitor, you are minimizing the power used already, as it is not doing anything else but video processing.