motioneye-project/motioneyeos #1333: Outdoor camera cases

StuartOswald, 2018-01-18:
Not really a question for the admin here but rather the community.

Could you recommend RPi2 and Zero W cases with or without IR functionality? What would be a good online store to find them?

Very interested to hear from anyone who has used them.

jasaw, 2018-01-18:
I am using this outdoor [dummy camera](https://www.aliexpress.com/item/Fake-Camera-Waterproof-Outdoor-Indoor-Security-Dummy-CCTV-Surveillance-Camera-Flashing-Rred-LED/32815686364.html) case with Pi Zero W inside. It's a tight fit, so I 3D printed mounting brackets to replace the internals.
[image]
[image]

mb39924, 2018-01-19:
I'planning on doing something like this
https://youtu.be/LPHJNXNyK9M

althought I do expect some temp and condensation problems..

classofoutcasts, 2018-01-24:
I'm using this one. Removed the analogue camera, added to small servo's for pan/tilt functionality to the plastic inside frame and with a few small modifications managed to mount the pi camera v2 in it. Works really well!
http://www.ebay.es/itm/Hiwatch-DS-2CE56C0T-IMM-HiWatch-HikvisionC%C3%A1mara-domo-HDTVIGama/182792942375?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2060353.m2749.l2649

fordorpi, 2018-02-01:
Look at the case used in this how-to
[https://www.raspberrypi-spy.co.uk/2017/04/raspberry-pi-cctv-camera-with-motioneyeos/](url)

jasaw, 2018-06-22:
If anyone is interested, I have my outdoor RPi Zero W camera hardware design here: https://github.com/jasaw/camera_hardware