motioneye-project/motioneyeos #1338: cannot write latest motioneyeos image to SD card

AhmedSalad, 2018-01-23:
hello,

i seem to be getting a "SD not formatted. format SD card before you can use it" popup when i write the latest raspberry pi image to my SD card. ive used different SD cards thinking it was that but to no avail. i also tried downloading the ZIP file but windows does not fine  an image once the file is unzipped.

maybe it has something to do with having to delete ".gz" off the img before windows can recognize motioneyeos as an image? help!

jasaw, 2018-01-23:
You can't just delete the .gz extension. You need to extract the file first.

AhmedSalad, 2018-01-23:
Even if i download just the image file? I inderstabd extracting the zip. Sorry if this is a terrible question. Im just stuck stuck. Lol

jasaw, 2018-01-24:
All the files available for download are gz compressed, so you'll need to extract the `img` file from it.
Here's a guide on how to extract a gz file: https://www.wikihow.com/Extract-a-Gz-File
Hope that helps.

AhmedSalad, 2018-01-24:
Awesome! Thank you!!!!

jasaw, 2018-02-02:
Issue appears to be resolved. Closing issue.

closed 2018-02-02 by jasaw