motioneye-project/motioneyeos #1339: complex password on SMTP settings removed all cameras

zaktalanos, 2018-01-24:
On setup on raspberry pi zero w, added long password for SMTP with aA1 charset and a ' character. pass is about 20 chars long.

This seems to have broken the os to the point where it assumed I had no cameras on the system. 

The only option to recover was to reload last good setup. 

Now I am using a simple alpha only charset and it is working and sending mails.

working
[image]

broken with old mail settings
[image]

I can provide more info if required.

ccrisan, 2018-01-24:
@zaktalanos I'll try to reproduce the issue myself. What is "aA1" charset?

zaktalanos, 2018-01-25:
small letters, capital letters and numbers. eg 252ThisIsthe'PASS

mb39924, 2018-01-26:
You can also try to create an app password

https://support.google.com/accounts/answer/185833?hl=en