motioneye-project/motioneyeos #1341: Access the binary motion images

andreburnier, 2018-01-24:
Hello,
I'm working on a artistic project that needs to track the motion of people on the camera, so I decided to test MotioneyeOs. I'm looking how to access - and save - the binary images of the motion tracking to get the silhouette of moving bodies. Something similar to this: https://youtu.be/nP6yerqf6eY?t=4m23s
I've tested the debug files option available, but they don't work as I intend.

I'm no programming expert, so I'm not sure where and how I could tweak the code to access this data. Any help is appreciated.

jasaw, 2018-01-24:
Motion detection is done by [motion](https://github.com/Motion-Project/motion) software, which is written in C. When and what and how images are recorded into videos are all handled by this software. Motion developers might be able to point you to the right direction.

ccrisan, 2018-01-26:
@andreburnier I'm not sure what your aim is here but if you want to access single raw images outside of motion(Eye), you have two options:

1) enable still images and either give a non-parametrized filename pattern to them (such as `last_image`) or rely on the `last_snap.jpg` symlink that motion generates
2) use some HTTP client such as `curl` to access the snapshot URL supplied by motionEye.

closed 2018-01-26 by ccrisan

andreburnier, 2018-01-29:
Hey @ccrisan @jasaw , thanks for your inputs.
I'll take a look into the Motion software.