motioneye-project/motioneyeos #1343: BUG: Setting surveilance user password overwrites admin password

careyer, 2018-01-25:
Hi Calin,

using the latest pre-release (20180101) I think I have discovered a bug:
Initially I only had an admin password set ... (i.e. the surveilance user had no password). I then decided to also set a password for the surveilance user. I typed it in an saved. (I did not touch the password for the admin (which was displayed asterixed-out: ******).

Now the password for 'user' (the newly entered one) works likea  charm. But the password for the admin user does not work anymore (Invalid Credentials). Seems lilke it has been overwritten as well with maybe some default stuff? Do you have any idea how I can gain back admin access?

Thanks!
Thomas

careyer, 2018-01-25:
**Update:** Here we go! In complete despair I tried admin/***** (user/pass) and it worked. I gained back admin access. Seems like there is a bug in the "General Settings" dialog. If only one of the two password fields is altered the other one gets saved as well with the asterixes as the new pass.

ccrisan, 2018-01-26:
Then this is definitely it's bug #1116. In short, clear your browser's cache, refresh and then update your password to whatever you want. You should be fine thereafter.

closed 2018-01-26 by ccrisan

careyer, 2018-01-26:
Sort of...
it seems like the GUI just updates both passwords admin and user password even though the admin password has not been changed/edited. It just writes the asterixed-out characters as a new password which is: *****