motioneye-project/motioneyeos #1347: FTP to upload video to NAS

HWL1968, 2018-01-28:
Hi all,

I am having an issue with uploading video files to a NAS. In MotionEyeOS (most recent version 20180101) I have configured to upload media files to a central NAS. Configuration is:
IP 192.168.1.xxx ( has an actual and valid number offcourse)
all 3 upload buttons are 'on'
upload service is 'FTP server'
server address is 192.168.1.xxx
port is 21
location is 'Surveillance' (without the ')
include subfolders is 'on'
username and password have been filled (and valid)

When I push 'test Service' nothing happens (no error message, no succeed). On my NAS nothing can be found either.

What log files do you need?

Please help.

Best regards.

ccrisan, 2018-01-28:
first Expert Settings > Enable Debugging, then motioneye.log should tell us more.

HWL1968, 2018-01-29:
Hi Calin,

Thx for your reaction. I enabled the debugging level and triggered some motion to record a video.

In the motioneye.log logging file, following is logged.

2018-01-29 18:17:23: [motioneye]    DEBUG: loading upload services state from "/data/etc/uploadservices.json"...
2018-01-29 18:17:23: [motioneye]    DEBUG: loaded upload service "ftp" for camera with id "1"
2018-01-29 18:17:23: [motioneye]    DEBUG: ftp: uploading file "/data/output/Zijkant_huis/20180129/18/20180129-181620-60.mkv" to ftp
2018-01-29 18:17:23: [motioneye]    DEBUG: ftp: size of "/data/output/Zijkant_huis/20180129/18/20180129-181620-60.mkv" is 18.098MB
2018-01-29 18:17:23: [motioneye]    DEBUG: ftp: mime type of "/data/output/Zijkant_huis/20180129/18/20180129-181620-60.mkv" is "video/x-matroska"
2018-01-29 18:17:23: [motioneye]    DEBUG: ftp: creating connection to FTPGast@192.168.1.11:21
2018-01-29 18:17:23: [motioneye]    DEBUG: ftp: ensuring path /Surveillance/20180129/18
2018-01-29 18:17:23: [motioneye]    ERROR: failed to upload file "/data/output/Zijkant_huis/20180129/18/20180129-181620-60.mkv" with service ftp: 550 Surveillance: File exists
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 875, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 98, in upload_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 650, in upload_data
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 703, in _make_dirs
  File "usr/lib/python2.7/ftplib.py", line 589, in mkd
  File "usr/lib/python2.7/ftplib.py", line 251, in sendcmd
  File "usr/lib/python2.7/ftplib.py", line 226, in getresp
error_perm: 550 Surveillance: File exists

I can assure you that the mkv does not exist.

Furthermore I tried sftp from the MotionEyeOS raspberry (ssh into RPi, then started sftp towards the NAS):

sftp -P 2222 <username>@192.168.1.xxx
Password:
Connected to 192.168.1.xxx.
sftp> pwd
Remote working directory: /
sftp> ls
Foscam         Home           Surveillance   TVRecording
sftp> cd Surveillance
sftp> ls
Voordeur         Zijkant_huis     logging_ftp.txt
sftp> mkdir test
sftp> ls
Voordeur         Zijkant_huis     logging_ftp.txt  test

As you can see, I can create a directory (also a file if needed) with the same username/password (therefor: rights have been set properly). 

Just a thought (didn't look at the Python files): are you trying to create the directory structure in 1 mkdir statement?

There is also an entry made in the google group forum. Which one would you like to keep?

Best regards.

ccrisan, 2018-01-30:
[This](https://github.com/ccrisan/motioneye/blob/master/motioneye/uploadservices.py#L691) is the function that takes care of creating the target directory on the FTP server.

Even though it's not my code (it was a contributed pull request), I see nothing wrong with that code.

Please note that the service you're using is FTP, while you tried to manually do stuff using SFTP. The two are fundamentally different. It might be worth trying that with ftp.

> There is also an entry made in the google group forum. Which one would you like to keep?

The group is for user discussions, while this is an issue tracker. As long as this is a valid issue (and it is, so far), I believe the github issue tracker is the best place for it.

HWL1968, 2018-01-30:
Hi,

Thx for your respons. I know that sftp is not what you were using (as stated in de logs it's the python script). I tried to use ftp on MotionEyeOS but did not succeed (not installed??). I could find ftp however, so tried that one (just to make sure that the NAS was reachable and accepted connections from the RPi).

What can I do to test further? As mentioned: the debug file lists 'File exists' but that is not so. (I double checked: from the RPi point of view as well as from the NAS). Am I able to test the separate python steps myself? (to see where the issue starts)

Have I configured anything wrong?

Best regards.

ccrisan, 2018-01-30:
You could try installing motionEye on Raspbian. On Raspbian you'll have more possibilities of debugging, including the `ftp` command.

HWL1968, 2018-01-30:
that will take more time than just half an hour ;-)

Will have a look if that can be done during the weekend. In what sequence do you start the python scripts?

ccrisan, 2018-01-30:
> In what sequence do you start the python scripts?

I'm not sure what you're asking here.

HWL1968, 2018-01-30:
I would try it step by step en see if I could find the point where it goes wrong. I assume that testing will be faster than installing raspbian and motioneye and test from there. Just looking to get more info where the script fails. My interpretation of the debug is:
- ftp setup is ok (connection is accepted by the NAS)
- but then? 'ensuring path'... is the directory being created? or is there no directory created?

 If I interpret the python script correctly (but I am no python programmer) the script tries to change directory and if not possible, it creates it (lines 691 - 707). It should create the correct directories, but it doesn't. Therefor I am wondering what is happening and how to debug any further.

HWL1968, 2018-02-01:
Hi,

I did install motioneye from scratch on a clean Raspbian image (on top of motion) as you describe in your tutorial: https://github.com/ccrisan/motioneye/wiki/Installation .

Still the same issue. What I notice however is: the RPi (motion?) places a '.keep' file in the directory when I declare it as a samba share afterwhich Motion can't delete any file.

How to proceed? I would like to use the FTP service.

HWL1968, 2018-02-03:
Hi Calin, as mentioned: I reinstalled the system as MotionEye on raspbian but the same issue still occurs. FTP can be done, files being created when using ftp myself (as the same user that motion does). I think I stick to motioneye as it is running right now. I have however the issue with the .keep file. I will open a new issue for that one.

closed 2018-02-03 by HWL1968