motioneye-project/motioneyeos #1348: Raspberry Pi - Root partition full - System keeps rebooting

ioCoffre, 2018-01-28:
Hi,

As the title says it, my MotionEyeOS keeps rebooting because the root partition (partition 2 of the microSD) is 100% full.
/lib uses 68 MB and /usr uses 91 MB (/usr/lib uses 72M)
Which files or folders can i remove safely in order to make my system work again?

Note: /data in partition 3 is only used at 35% and partition 1 (FAT32) only 42%

Thank you.

ccrisan, 2018-01-29:
1. How do you know that the cause of reboot is the root partition?
2. The root partition is supposed to be read-only and never to be written to. Did you remount it read-write and do stuff on it? This may happen if you enabled debugging for a long time.
3. You can't and you shouldn't remove anything on the root partition.

ioCoffre, 2018-01-29:
Thanks for your reply.

1. It's an assumption based on other issues which were having the same message I got "No space left on device" with a system always rebooting.

2. I did not do stuff on the root partition. I only mounted it on my Debian server to see which partition was out of space. And it happened to be the root one.

3. Duly noted, that what I was thinking too. So I expanded the root partition (and reduced the  data partition) with GParted, but now the system doesn't seem to boot anymore. I'm gonna reinstall it.

Is there a way to install MotionEyeOS with a root partition bigger than 180 MB ?

ccrisan, 2018-01-30:
> It's an assumption based on other issues which were showing the same message I got "No space left on device" with a system continuously rebooting.

I need to see some logs saying that.

> Is there a way to install MotionEyeOS with a root partition bigger than 180 MB ?

No, there's no (easy) way to do that and there's no point in doing that. I repeat: **the root partition is read-only** and should never be written to. Therefore the remaining disk space should be irrelevant.

ioCoffre, 2018-01-30:
I won't be able to provide any log now as I have reinstalled everything. But trust me, the system was starting to boot, then it showed that out of space message, after that all services were shutting down, then it said system is rebooting and it was cycling like that boot after boot.

OK, root partition is read only, I got it the first time. I don't know if it is used at 100% by default but that's what du -h was saying.

cloudlyfe-aldwin, 2018-02-02:
I have a failed unit that I may be able to pull the logs from. What is the location of the logs that you need?

ccrisan, 2018-02-05:
@cloudlyfe-aldwin do you experience the same problem as @ioCoffre? If yes, i'd need to see the log files from `/var/log` (`messages`, `dmesg.log`, `motioneye.log`, `boot.log`)

msq-ie, 2018-02-23:
@ccrisan I got this issue:
```
[root@meye-528cb2b1 ~]# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               174.3M    167.7M         0 100% /
devtmpfs                433.4M         0    433.4M   0% /dev
tmpfs                   437.9M     16.0K    437.9M   0% /dev/shm
tmpfs                   437.9M    148.0K    437.7M   0% /tmp
/dev/mmcblk0p1           19.9M      8.5M     11.5M  42% /boot
/dev/mmcblk0p3           13.9G     61.8M     13.2G   0% /data
/dev/mmcblk0p3           13.9G     61.8M     13.2G   0% /home/ftp/sdcard
/dev/mmcblk0p3           13.9G     61.8M     13.2G   0% /home/ftp/storage
/dev/sda1               342.8G     43.8G    281.6G  13% /data/media/sda1
/dev/sda1               342.8G     43.8G    281.6G  13% /home/ftp/storage/sda1
```

And looks like I can still access the /var/log or as it is linked /data/log. Let me know if you want those files. And does it make a difference if I will get the via ssh vs via front end?

ccrisan, 2018-02-24:
@msq-ie what is your exact issue? From what I can see, your `df` looks good. There's no need to worry about `/dev/root` as it's read-only and will never be written to.

msq-ie, 2018-02-24:
@ccrisan System being unresponsive from time to time. And by unresponsive I mean no ssh or web interface, not even responding to ping.
Sometimes it is once every few days, sometimes few times a day. One usb camera and WD Pi drive connected on separate power supplies. And I'm not using motion detection - just take picture every 2 seconds, enabled manually every morning.

ccrisan, 2018-02-24:
@msq-ie this sounds like a totally different issue. You could attach your logs here (**not via email**), but I can't promise anything. You might be better off "debugging" your problem by removing/replacing one component at a time from your setup.

ioCoffre, 2018-04-27:
Hi,
It's doing it again nearly 3 months after a fresh install on a SanDisk Ultra 16 GB card.
http://jomay.free.fr/01.mp4
I saw on another thread that you said the solution is to mount the µSDcard on a Linux and delete some recorded files.
Will it be possible to have an option that prevent the card to be fully filled so it will still allow us to boot up and delete from the web interface _or_ an option that delete automatically older files to free space for newer ones?
Thanks for any reply and have a nice day.
io

ioCoffre, 2018-05-13:
So I will reinstall again...

jasaw, 2018-05-14:
@ioCoffre I saw your original post, and looks like your data partition (partition 3) was only 35% full, so your system rebooting was not caused by disk being full (I don't know what the cause it yet). Seeing that you've got the same problem 3 months later, it's probably the same problem as before. Can you please check how much free space left on your data partition (partition 3)?

I have one theory: if your data partition is corrupted, it gets mounted as read-only, and the system is unable to write to syslog, so you get a disk full message. Did you use the same SD card as before? Did your device lose power prior to this rebooting problem?