motioneye-project/motioneyeos #1349: Upgrade to v20171003 breaks USB webcam

mmeah, 2018-01-29:
Upgrading version 20170710 to 20171003 breaks USB Logitech HD Pro Webcam C920

ccrisan, 2018-01-29:
Have you tried more recent versions, such as 20180101 or 20171008?

mmeah, 2018-01-29:
@ccrisan Upgrade to 20171008 shows the same error. The other wifi camera still works fine.

[image]

mmeah, 2018-01-29:
[root@meye-d5c8063a ~]# fwupdate upgrade 20180101
upgrading to 20180101
downloading...
no such version

mmeah, 2018-01-29:
Looking at motion.log shows this:
[2:ml2] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.86.90:88/videoMain)
[1:ml1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[2:nc2] [ALR] [NET] netcam_handler_loop: Camera handler thread [4] started

ccrisan, 2018-01-30:
> [root@meye-d5c8063a ~]# fwupdate upgrade 20180101
> upgrading to 20180101
> downloading...
> no such version

You need to enable prereleases (in expert settings) for that.

> [2:ml2] [ALR] [NET] netcam_start: Network Camera thread starting... for url 

I thought you had problem with a USB webcam. So is it a network camera or a USB camera?

mmeah, 2018-01-30:
Ah, yes it downloads now after turning on prerelease.

[root@meye-d5c8063a ~]# fwupdate upgrade 20180101
upgrading to 20180101
downloading...
downloaded 20180101
extracting...
extracted 20180101
flashing boot...
boot ready 20180101
preparing for reboot...

And it works!

I have both USB and wireless, the USB device broke.
Only 20171003 and 20171008 have the breaks.

Thanks @ccrisan for the hint. You the man!

closed 2018-01-30 by mmeah