motioneye-project/motioneyeos #1351: Feature Request: Automatically create / email timelapse

pkwentus, 2018-01-29:
I find that I am manually repeating the same steps every day: Creating a timelapse to check for activity then reviewing each camera. 

I'd love for this timelapse to be created automatically so that I can quickly review to see IF anything happened while away.  I'd really like to push this timelapse to google drive / email so that  I can review at my leisure.

Any other builds  on this idea??

Felixrising, 2018-01-29:
So like an intervalometer of each camera throughout the day? Or do you mean a timelapse of the captured still images created on each event? Both would seem useful.

pkwentus, 2018-01-30:
I get quite a few pics during the day - ranging from 100 - 250 per camera.  Mainly, this is due to wildlife, leaves and shadows.  I've played with settings and have just learned to live with all of the pictures.

To speed up the review process, I think it would be great to have a daily summary- just a mashup of all of the images collected by the camera.  I'd like to push this off of the Pi either using Email, Google drive or drop box - just so it is easier to review.

Felixrising, 2018-01-31:
I have my camera likewise setup on the side of too sensitive (rather than not sensitive enough) and also have them pushing to google drive where I review them using the internal viewer (both videos/photos).
I'd rather see MotionEye move to supporting something more intelligent like OpenCV which can do much better image recognition (e.g. identify humans/faces or whatever you train it for)... Motion itself is really not that great, and barely under active development at all.

pkwentus, 2018-01-31:
I agree that motion has flaws.....   the argument that I fell is most strong is that MOTION MISSES EVENTS.  I've dialed all settings towards sensitivity and still ~15% of the time miss my car driving up in the afternoon.  I do have lots of pictures of suspected motion of gravel though ;)