motioneye-project/motioneyeos #1352: Pi 1 Model B, keeps rebooting, eth0 has no ip address

jameselsey, 2018-01-30:
Hi,

I'm using `motioneyeos-raspberrypi-20180101.img` on a Pi 1 B.

It's connected via ethernet directly to my home router, the wire works, I've plugged in another device and it connects fine.

The motioneye pi seems to be stuck in a repeating cycle of rebooting. Sometimes it pauses on the login prompt and gives me enough time to hit the UI and see camera footage, but then it reboots again.

The startup log says it's assigned `192.168.1.5` via DHCP, which is the address I'm using in browser.

I've checked my router, it shows up under DHCP clients:
```

Hostname | MAC Address | IP Address | Expires In
-- | -- | -- | --
raspberrypi | b8:27:eb:7d:0e:d2 | 192.168.1.2 | 21 hours, 32 minutes, 22 seconds
Jamess-MBP | 8c:85:90:2a:b9:94 | 192.168.1.3 | 22 hours, 42 minutes, 6 seconds
unknown host | ec:71:db:ee:51:72 | 192.168.1.4 | 18 hours, 51 minutes, 24 seconds
meye-90abeef0 | b8:27:eb:ab:ee:f0 | 192.168.1.5 | 23 hours, 59 minutes, 5 seconds
```

Also checked the LAN stats on the router

```
Interface | Received | Transmitted
-- | -- | --
  | Bytes | Pkts | Errs | Drops | Bytes | Pkts | Errs | Drops
eth1 | 4077375 | 3551 | 0 | 0 | 2517178 | 34024 | 0 | 0
eth2 | 1930032 | 30132 | 0 | 0 | 14145707 | 66811 | 0 | 0
eth3 | 371873 | 3262 | 0 | 0 | 15908312 | 95913 | 0 | 0
eth4 | 0 | 0 | 0 | 0 | 2146309 | 31237 | 0 | 0
wlan | 7586699 | 48656 | 0 | 0 | 42615152 | 84792 | 0 | 0
```

[image]

What else can I try? I've searched other issues like this but haven't found a resolution

ccrisan, 2018-01-30:
Power issues perhaps? Once the ethernet interface goes down, it loses its IP address.

jameselsey, 2018-01-30:
I don't think it'd be power issues, I'm using a USB mains cable, but I can try another USB plug just to be sure.

Are there any logs I can find extra details?

ccrisan, 2018-01-31:
`/var/log/messages` and `/var/log/dmesg.log` could tell you more about what's going on.