motioneye-project/motioneyeos #1354: No attachments in notification emails

mireksochor, 2018-01-30:
No attachments in notification emails :(
- Attached Pictures Time Span: 5
- Capture Mode: motion triggered (one picture)

motionEye Version | 0.38
Motion Version | 4.1
OS Version | Raspbian 9.1
Raspberry Pi B

apologies, I am new here:)

ccrisan, 2018-01-31:
Have you checked https://github.com/ccrisan/motioneyeos/wiki/FAQ#what-should-i-do-to-have-pictures-attached-to-motion-notification-emails?

seb276, 2018-07-12:
I install the latest version motioneye on Raspi3 and I have got mails with attachments with "motion triggered" but not with "motion triggered one picture".

Do you have and idea?

Thanks a lot

ccrisan, 2018-07-12:
https://github.com/ccrisan/motioneyeos/wiki/FAQ#what-should-i-do-to-have-pictures-attached-to-motion-notification-emails

closed 2018-07-12 by ccrisan

seb276, 2018-07-12:
Thanks, I read FAQ before and I configure Attached Pictures Time Span to a reasonable amount of seconds (30).

"motion triggered one picture" is working for you?