motioneye-project/motioneyeos #1355: PI2 Unable to open video source after update

KiwiNibbler, 2018-01-30:
Hi all,

I had a dead camera for a couple of months.  Yesterday I decided to rebuild it and get it back online.  All good.  But I noticed that there was a new version available (I ticked the box to get pre-releases).  So I upgraded the camera I rebuilt.  All good.

I decided to update my other two as well.  Half good.  One of them is fine; the other now shows the grey screen of death.  Camera two and three are identical in every way.  Same IR Camera (Which is why the colors may seem slightly off on camera 2).  But camera 3 did not like the upgrade.

All three cameras have the exact same configuration with the exception of camera names and IP addresses obviously.  A bit disappointing since camera 3 had been my most reliable historically.

[image]

I have logs but would prefer to not post them publicly.

ccrisan, 2018-01-31:
Without logs (specifically motioneye.log) I won't be able to help you.

KiwiNibbler, 2018-01-31:
Hi there,

I wasn't sure which log files would be most useful.  Here is my motioneye.log.  Debugging is on.  

Before grabbing the log, I did a reboot.  I also checked disk usage.  It has used .5 Gb of 14.5 Gb, so no issue on free space.
[Camera 3 Motioneye.zip](https://github.com/ccrisan/motioneyeos/files/1683114/Camera.3.Motioneye.zip)

Any insight would be greatly appreciated.

KiwiNibbler, 2018-01-31:
It just occured to me that camera 2 and 3 have similar cameras, but not identical.  They used to be identical but the IR part of camera 3 died and I replaced the whole thing about six months ago.  It is a slightly different version, although nearly identical.  If that makes a difference.

ccrisan, 2018-02-01:
Do you have 3 different Raspberry PIs? Or do you have a single one with 3 cameras connected locally? You mentioned updating 3 cameras and I assumed you meant 3 Raspberry PIs. Now you have sent me a motionEye log file that shows multiple cameras connected to same board. This doesn't make too much sense.

KiwiNibbler, 2018-02-01:
I have three different PIs.  I inadvertantly sent you the logs from the headend, not the camera that is having issues.  I'll get a new log when I get home for just that one camera and post it.

KiwiNibbler, 2018-02-02:
I did a hard reboot (turned off - turned back on) the faulty PI.  It is fine now.  I didn't realize that when I was getting logs from a head-end device, even if I have selected one of the other devices and clicked settings for it, it gets the logs for the head-end.

closed 2018-02-02 by KiwiNibbler