motioneye-project/motioneyeos #1357: Motion photos taken too late

fordorpi, 2018-01-31:
I have a pi zero set up with the zeroview and the camera is mounted on a window pointed toward my driveway.  All of the photos that are sent to my email are either too late or way to early.  I do not get a car pulling out of the garage or a person when someone walks by the window....just a nice photo of my driveway.
Is there any documentation on how to set up the motion capture to get photos with the motion shown on them?

mb39924, 2018-02-02:
I have these settings which work great.
[image]

For the notification I set Attached Pictures Time Span to 4

Felixrising, 2018-02-03:
What is your video device frame rate set to? That could have a big impact on the "minimum motion frames" setting and is an important part of the motion detection trigger.

fordorpi, 2018-02-04:
Thanks mb39924.  I made the changes and I will see what happens tomorrow.

Hooch76, 2018-02-06:
> I do not get a car pulling out of the garage or a person when someone walks by the window....just a nice photo of my driveway.

Add me to the list, too. The same is happening for me, too. 
It would really be helpful if there was a description that would explain all that options. I was not able to find one, e.g.

- Motion detection: Captured before / Captured after: is this only applicable for creating the movie or does it also apply to pictures?

- Still images: Capture mode: Motion triggered (One Picture), Motion triggered, All frames. What's the exact difference?

Frankthedoc, 2018-02-22:
Not sure if this applies to you, but I'll post as it may answer for some people: I had this problem when I was using an older Pi (B+) with a 1080p camera. Delay in capturing an image could be several minutes! My resolution was to buy an ODroid XU4, which can handle all the data coming its way (motion does a lot of CPU work). An alternative might be to use a lower resolution output from the camera (if available). Masking or dropping the resolution in motioneyeos itself didn't yield good results for me.