motioneye-project/motioneyeos #1360: Server running well, unfortunately... camera displaying grey

Daneddins, 2018-02-03:
Calin, 

Im am EXTREMELY new to coding. I told myself this year that I would learn the basics. So far so good. I decided to use your tutorial to help me get into it and work on my next project. I have the Raspberry Pi Zero W running raspbian stretch lite, and motioneye from their most current repositories. I enabled the camera, ssh, and have it running properly. I checked to see that the camera was working before I installed motioneye by pulling an image from the camera module. I set everything up, and finally got the apache server to broadcast over my Wifi, but for some reason when I boot the front end, It displays my camera as a grey background. Any suggestions would be greatly appreciated!

Thank you kindly for helping beginners like me!

Dan

[image]

ccrisan, 2018-02-05:
What camera type/model? What board? What OS version? Is this reproducible?

ccrisan, 2018-02-05:
> What camera type/model? What board? What OS version? Is this reproducible?

Nevermind, I thought this was really about motionEyeOS. After reading your description more carefully, I realized it's about plain motionEye + Raspbian on a RPi Zero W.

ccrisan, 2018-02-05:
Since you set up apache as a reverse proxy (I assume), does motionEye work fine standalone?

Daneddins, 2018-02-05:
Mr. Crisan, 

I am running the Raspi Zero W with Raspian Stretch and motioneye downloaded per the instructions in your wiki. MotionEye works great and is functioning well, with the exception of the grey output of the camera. No issues with logging in or finding the camera on the board.

Here are some text files. is there anything else from the boot file that would help?

[cmdline.txt](https://github.com/ccrisan/motioneyeos/files/1696773/cmdline.txt)
[config.txt](https://github.com/ccrisan/motioneyeos/files/1696774/config.txt)
[issue.txt](https://github.com/ccrisan/motioneyeos/files/1696775/issue.txt)

Kind Regards, 

Dan

ccrisan, 2018-02-06:
@Daneddins I need to see your motioneye log file. You can obtain it by running:

    journalctl -u motioneye > motioneye.log

Daneddins, 2018-02-07:
Mr. Crisan, 

Please disregard my earlier post from this evening. As I has a feeling I made a stupid mistake by making a directory file out of the video0. I decided to uninstall motion, disabled/reenabled the camera module in `raspi-config` rebooted and it is now broadcasting. 

Thank you for your contributions, answering questions, and for maintaining motioneye. 

Kind Regards, 

Dan

closed 2018-02-07 by Daneddins

JulianMHall, 2020-04-13:
Hi,
I have the same problem with the grey screens albeit I am not running a server. Motioneye is installed on a Pi2 with a WiFi dongle and does function, except for this issue.
Motioneye works properly with the beta version phone app and the cameras all display with the app that came with them. They are standalone wifi cameras which deliver an RTSP url to Motioneye. It worked fine for a long time but has now stopped working with no configuration changes. I have tried reinstalling and reloading from a saved config file with no change. I then reinstalled and ignored the config file in case it was corrupt with no improvement. The cameras display properly if I access the urls directly so the browser is fine.
I tried to extract the motioneye log file as you requested above, but that gave an error saying it is a read-only file system (I had logged in as root).
Kind regards,
Julian

starbasessd, 2020-04-13:
Can you describe you r setup a little more, please?
Which version of MotionEye or motionEyeOS are you running on the PI2?
You say you are not running a server? But all cameras are being read by the
Pi2 and fed to an app?
Can you connect the browser on the phone to the Pi2 running motionEye?
Either port 80, or port 8765 if running motionEye in Raspbian

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JulianMHall, 2020-04-13:
Thanks for replying.
Copied from Motioneye screen.
motionEye Version | 0.39.3
Motion Version | 4.1.1
OS Version | motionEyeOS 20190119
The OS is installed on a microSD card in a Pi2, the Pi2 gets an IP from my router. I then access that IP with my browser. It /was/ working fine but now the cameras don't display.
I meant that I am not running Apache etc as the original poster was.
The Motioneye app is apparently a beta version, but that does display the cameras. The app that they came with (as I said they are standalone Wifi cameras) also sees them perfectly.
I /can/ connect the phone's browser to Motioneye, and I was prior to finding the Motioneye app (available from Google Play for Android, and probably the iPhone's App Store).
Incidentally how do you update the OS? I have tried the Update option within the OS and it claims to update but the version never changes.
Kind regards,
Julian

starbasessd, 2020-04-13:
Update: Expert Settings: Turn on Enable Pre-Releases, then do check and
update.
or
** My Prefered ** Download and install your Pi Version from here:
https://github.com/ccrisan/motioneyeos/releases/tag/20200203
If you have a PC, can you connect to http://<ipaddress> and get the cameras?
Are the gray windows only on the browser on the phone?
Some people have reported that they were getting the spinning gray screen
for cameras with the latest Firefox (it happened last year  too, I
believe), on somewhat older motionEyeOS but upgrading motionEyeOS cleared
it up.
I would try a different browser, too, like IE, Edge, or Chrome

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

JulianMHall, 2020-04-13:
I will install the way you suggested. The first option as usual took a long time then just displayed the same version information as I already pasted, so seemingly did nothing.
Yes, as I posted above, in the same browser if I enter the url for any of the cameras I can see them.
The /only/ software which will not display them anymore is Motioneye OS.
I'll try other browsers and updating Motioneye again.
Kind regards,
Julian

JulianMHall, 2020-04-14:
OK that I do not understand. Updating the OS made no difference, nor did removing the cameras and re-adding them. Trying in IE made no difference. However, /adding/ one in IE worked! Not only that but after adding one in IE I was able to add them in Firefox!
That makes no sense at all, but it works now so I'm not questioning it.
Many thanks for your help!
Kind regards,
Julian

eekie76, 2023-04-10:
That did not work for me. I am using chrome and tried ie and edge but the problem remained.
For several years I had no problem putting motioeyeos and the supplicant file on an sd card with Raspberry Pi 0 w and v1.3 camera, but now with newly bought camera's it works once but after reconnecting the camera flat cable I get the gray screen with hardly readable text: unable to ......
I can barely believe I wrecked the camera's. There must be a software problem. Please help!

starbasessd, 2023-04-10:
"Latest version" is [dev20201026](https://github.com/motioneye-project/motioneyeos/releases/download/nightly-dev/motioneyeos-raspberrypi-dev20201026.img.xz)
Check the camera with the command:
> vcgencmd get_camera 
> cd /data/output 
> raspistill -o output.jpg

if you get an error like:
> mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates

run the commands:
> service motioneye stop
> raspistill -o output.jpg