motioneye-project/motioneyeos #1362: Wrong resolution if rotated

KarlBaumann, 2018-02-05:
Hi!

First of all, thanks for the amazing software!

My camera is rotated 90deg. and it means all available resolutions should adapt to that rotation. 
Currently I am getting cropped image with all available resolutions.

ccrisan, 2018-02-05:
Can you be more clear about what your situation/problem is? Is your camera physically rotated? Doesn't the rotate configuration option help? Screenshots may be useful here.

KarlBaumann, 2018-02-05:
Yes, my camera is physically rotated, so I use option in the configuration and set it to 90 degrees.
After I do that, Image rotates to the correct angle, but gets cropped (I see only middle part of it).

I will add some screenshots later, if necessary

KarlBaumann, 2018-02-05:
So I tried to make some screenshots and suddenly all problems are gone.
I guess it was a temporary issue.

But thanks for the instant reply!

KarlBaumann, 2018-02-05:
Now I finally understood how to reproduce it: "Fast Network Camera" mode has to be enabled.
[image]

TomThorpe, 2020-04-09:
Hi,

I’m having this same issue, and have the same repro steps as above. If the camera is rotated 90degrees when “fast network camera” mode is enabled, the video ends up being the wrong aspect ratio (it remains the original aspect ratio not the rotated one) meaning the video gets severely cropped.

lcnittl, 2021-08-23:
Same here, in fact it seems that the resolution does not get translated for `90°`/`270°` when `fast network camera` is enabled.

Example: Resolution is set to `1280x960` and rotate is set to `90°`. While the video is correctly rotated, the stream still has a resolution of `1280x960` and not the expected `960x1280`, which leads to a center cropped and enlarged image like this one:

[image]

starbasessd, 2021-08-23:
And what happens if you modify the resolution to 960x1280 manually?

lcnittl, 2021-08-23:
You mean in the config file? (The GUI does not allow manual entry, at least in my case)

starbasessd, 2021-08-23:
In Video Resolution dropdown, at the bottom, Select Custom, then enter the correct aspect ratio of 980x1280

lcnittl, 2021-08-23:
Hmm - unfortunately, `Select custom` is not listed. Maybe due to the fact that I am using `RTSP` mode? In any case, I was able to get the desired effect by altering resolution settings in both `/data/etc/raspimjpeg.conf` and `/data/etc/motioneye.conf`. Thanks for the thought-provoking impulse 😄