motioneye-project/motioneyeos #1365: motioneye OS received interrupt signal and is shutting down

obihh, 2018-02-05:
Hello, i have the Problem that the Motionese OS shuts down by itself.   
After a new boot it sometimes takes two days, sometimes it shuts down in a few minutes.  In the Motioneye.log there are some "connection errors", after that the system shut down.

Where is the problem?  A websearch was without success...

Thanks a lot!

[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/1694928/motioneye.1.log)

marhau, 2019-03-03:
Same here. No idea why it happens and no consistent uptime between each time.

originalbleak, 2019-08-03:
Same here. I have two RPi Zero cams and, after a month of trials, finally got one stabalized. Not holding my breath on that but the OTHER one keeps shutting down with...

> 2019-08-01 16:23:53: [motioneye]     INFO: interrupt signal received, shutting down...
2019-08-01 16:23:53: [motioneye]     INFO: server stopped
2019-08-01 16:23:53: [motioneye]     INFO: tasks stopped
2019-08-01 16:23:55: [motioneye]     INFO: motion stopped
2019-08-01 16:23:55: [motioneye]     INFO: smb mounts stopped
2019-08-01 16:23:55: [motioneye]     INFO: bye!
2019-08-01 16:26:09: [motioneye]     INFO: hello! this is motionEye server 0.40
2019-08-01 16:26:13: [motioneye]     INFO: cleanup started
2019-08-01 16:26:13: [motioneye]     INFO: wsswitch started
2019-08-01 16:26:13: [motioneye]     INFO: tasks started
2019-08-01 16:26:13: [motioneye]     INFO: mjpg client garbage collector started
2019-08-01 16:26:13: [motioneye]     INFO: smb mounts started
2019-08-01 16:26:13: [motioneye]     INFO: server started

I'm using the latest stable 20190427. I thought the problem may have been wifi range because it's in the back of my house and the router is in the middle so I connected it to a NETGEAR Wi-Fi Range Extender which is in the same room but the problem still persists like it it did when it was connected to the router's wifi.

I don't notice any correlation to the crashes and other activity on my LAN.

jasaw, 2019-08-05:
MotionEye process receives interrupt signal and shuts down because another process is telling it to. I would look at wifi connectivity problem. Try increasing "Network Link Timeout" to something bigger like 300 seconds to deal with intermittent wifi dropouts.