motioneye-project/motioneyeos #1367: Static IP not found after changed in Panel

gbakakos1, 2018-02-05:
Having an issue with finding my static IP.   I get a DHCP address and all work fine.  I'm attempting to change that IP to static using the panel.  Once changed I get no found. Tried multiple Pi Zero w boards.

Original IP was 192.168.0.44
Changed that to 192.168.0.10

Seems simple enough...

[image]

AndyMAu, 2018-02-06:
Don't quote me as a networking expert.In my setup I use default gateway 192.168.1.254 and same for DNS. Works for me.

mb39924, 2018-02-06:
@gbakakos1 What exactly are you trying to do?

If I understand you correctly you want to set a static IP address but I don't get the changing part..

If you want to set a static IP address
- Find out the IP address of your installation. Put that address after "IP Address"
- Go into the control panel of your router en reserve that address to the mac address (you can find this info on your android phone using the app "Fing")
- Find out your Default gateway. You can find this in the control panel of your router or, if you are on a windows machine: Win+x > Windows Powershell > after prompt type: ipconfig
- Put this address after "Default Gateway"

(more details on finding default gateway here: https://www.lifewire.com/how-to-find-your-default-gateway-ip-address-2626072)

For most use cases you can leave Network Mask and DNS Server as is.

ccrisan, 2018-02-07:
@gbakakos1 what do you mean by "Once changed I get no found"? How do you not find it?

gbakakos1, 2018-02-07:
I'm sorry for not being clear.   I was unable to locate the new changed IP on my WiFi.   I found out what the issue was.  I have my main cable router plus two other additional routers setup.  Use tohave them on their own subnet.  I changed that to bridge them to the main router.   I incorrectly set that up.  Fixed that and of course its now working great.  Thanks to all for responding.

closed 2018-02-07 by gbakakos1