motioneye-project/motioneyeos #1368: Feature Request: Add GPIO functionality to control e.g. Relais/Solenoids with Motion/Timer

careyer, 2018-02-06:
Hi Calin,

would you consider adding control over GPIO pins in a future release? It would be awesome to be able to control GIPO pins depending on events such as: Motion detected, timer, etc...

I am using motioneyeos in a usage scenario were controlling GPIOs / Relais upon detected makes so much sense. Most 4 Relais breakout boards for raspberry use Pins 24,24+28,29 to control relais:
[image]

Thank you in advance!
Keep up the terrific work!

ccrisan, 2018-02-07:
Support for controlling GPIO is already present. You have the `RPi.GPIO` python module, you have a `gpio.sh` convenience shell script and you have raw access to `/sys/class/gpio`.

On the motionEye side, you have options to run a command on various events (such as motion detection). You also have [action buttons](https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons). It's up to you to write a script that glues the two together and implements your own logic.

I believe this is enough to help you implement your automation.

closed 2018-02-07 by ccrisan

careyer, 2018-02-07:
Thank you very much! Sorry for bugging you ... I did not realize that support for this was already build in. Your explanation really helps so much! Take care and keep up the excellent work.