motioneye-project/motioneyeos #1371: Adding Pi Camera - Crash

StuartOswald, 2018-02-09:
Using a Pi Zero W with Pi NoIR Camera V2 (tried multiple PSU). Build "motioneyeos-raspberrypi-20180101".

The systems become unresponsive after adding the camera in OS.

No logs to share as I can only cold boot and repeat until crash.

Any advice or suggestions greatly appreciated.

jasaw, 2018-02-09:
Try reflashing your SD card?

StuartOswald, 2018-02-09:
Thanks, jasaw, I have done that a few times and used different cards.

Just to be clear. As soon as I add the detected Pi NoIR Camera V2 I see a few slow frames before it freezes. I can use the UI but attempting to save and/or reboot eventually asks me to refresh.

jasaw, 2018-02-09:
Sounds like wifi issue to me. Have you tried running the Pi next to your wifi access point? Just to rule out wifi signal issue.

StuartOswald, 2018-02-09:
Thanks for the suggestion.

I am doing all this on the same table as my router.

Might the pi not have the processing power to handle this? Although I have
seen plnety of vids from others having no problems.

jasaw, 2018-02-09:
I have multiple Pi Zero W boards running Pi camera, and they have been working well for me so far.
Have you tried a different microUSB cable?
Did you overclock your Pi?
Your Pi is not inside a metal box, is it?

Check whether Pi is under voltage or throttled.
```bash
vcgencmd get_throttled
# 0: under-voltage
# 1: arm frequency capped
# 2: currently throttled
# 16: under-voltage has occurred
# 17: arm frequency capped has occurred
# 18: throttling has occurred
```

StuartOswald, 2018-02-09:
Thanks again,

I would love to have your pi zero setups. I've tried another USB cable and haven't overclocked it or put it in a metal case.

I managed to get the log files before it crashes. You can view them here https://drive.google.com/drive/folders/1f7EB2qHxehNLB3eVoU-ePj3q8PN08jwB?usp=sharing. Does this shed any light on the issue?

StuartOswald, 2018-02-09:
Running `vcgencmd get_throttled`, gives `throttled=0x0`.

jasaw, 2018-02-12:
motioneye.log shows an exception. Don't know if it's related.

>2018-02-09 15:07:41: [motioneye]     INFO: hello! this is motionEye server 0.38
2018-02-09 15:07:45: [motioneye]     INFO: cleanup started
2018-02-09 15:07:45: [motioneye]     INFO: wsswitch started
2018-02-09 15:07:45: [motioneye]     INFO: tasks started
2018-02-09 15:07:45: [motioneye]     INFO: mjpg client garbage collector started
2018-02-09 15:07:45: [motioneye]     INFO: smb mounts started
2018-02-09 15:07:46: [motioneye]     INFO: server started
2018-02-09 15:09:25: [motioneye]    ERROR: Exception in callback <functools.partial object at 0xb5d56570>
Traceback (most recent call last):
  File "usr/lib/python2.7/site-packages/tornado/ioloop.py", line 605, in _run_callback
  File "usr/lib/python2.7/site-packages/tornado/stack_context.py", line 276, in null_wrapper
SystemError: Objects/cellobject.c:24: bad argument to internal function
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                  2018-02-09 15:13:43: [motioneye]     INFO: hello! this is motionEye server 0.38

StuartOswald, 2018-02-16:
How would I investigate this further?

sfgabe, 2018-02-21:
I had a similar problem when adding a new camera onto a Pi Zero W. Try reflashing with the previous stable version of motioneyeos. I was able to get it going, recognizing the camera, and running with the older version for a bit, then used the auto update to move up to the latest version.

StuartOswald, 2018-02-21:
Thanks for the suggestion @sfgabe. I'm going to give this a try and will report back.