motioneye-project/motioneyeos #1373: February 2018 - Nonetype lower error still around.

Ryland0, 2018-02-09:
motionEye Version | 0.38
Motion Version | 4.1
OS Version | Raspbian 9.3

Trying to connect four Rpi Zero W cameras to my WD NAS storage device. This is not a fresh build on the cameras themselves, but I'm using a Rpi3 as the "controller" so I can access all four cameras from the web at once. I have the NAS mounted on the "controller" with no user or password and no issue. However, through the GUI I'm getting the same authentication error I've seen many times posted. Here is the output from my motioneye.log on pi-cam1.

2018-02-09 14:33:46: [motioneye]    ERROR: access to network share //10.8.24.7/Public/Motion/cam1 failed: 'NoneType' object has no attribute 'lower'

The odd thing is, I just rebuilt my "controller" and a few days ago it was working fine. I followed the same installation steps as I did previously following this guide.

https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

On the GUI, under File Storage, I have configured:
Storage Device - Network Share
Networks Server - 10.8.24.7
Share Name - Public/Motion/cam1 (also cam2 cam3 and cam4 on the others)
Share User - [blank]
Share Password - [blank]
Root Directory - /

There is no user or password on the public share.

Thanks!

Ryland0, 2018-02-09:
Also this from another suggestion.

[root@pi-cam1 ~]# mount.cifs //10.8.24.7/Public/Motion/cam1 /data/media/motioneye_10_8_24_7_public_motion_cam1 -o guest
mount error(16): Device or resource busy
Refer to the mount.cifs(8) manual page (e.g. man mount.cifs)

The data folder is currently collect photos. I set it to manual instead of motion trigger but same error. I am also unable to hit apply on anything other GUI settings as it knocks the camera offline from this storage entry.

Love this program by the way. Well done!

Ryland0, 2018-02-09:
Okay, I applied all the other settings I needed/wanted. I hit Apply and let all four cameras go offline. I closed the GUI, rebooted each and ran the above mount command again and it works! My photos are now on the NAS as expected. Not sure why the GUI was mad.

ccrisan, 2018-02-09:
The empty username NoneType error has been recently fixed in motionEye [0.38.1](https://github.com/ccrisan/motioneye/releases/tag/0.38.1). It'll be part of the next stable release.

closed 2018-02-09 by ccrisan

Ryland0, 2018-02-11:
Thank you for sending that link. I really hate to trouble you, but I can't
seem to get these files updated. I moved them over to the /home/ftp/storage
folder (the only one I could sftp to and write stuff, but trying to do what
I thought was the update, copying the files in the motioneye directory,
only gives me "read only file system". If there is a link to updating, I
will gladly follow it.

Thank you very much!

Ryland0, 2018-02-11:
Scratch that, I can wait until you release the newer version.

I have a feature request though. As I mentioned in my post, I have a Pi3
acting as my "controller" so I can view all four cameras at once, at home
and from the web. It would be awesome if this controller, once I've added
my cameras, had all the features available as from the camera GUI itself.
Like timezone control, reboot, etc. Also, if this controller could check
for updates for itself and also push updates out to any cameras it is
monitoring. That would be awesome. I wonder if having this as a poll on
your site would be worthwhile. It would seem likely that a lot of folks are
running a controller as I am.

Anyway, thanks again for your time. This is an incredible project and I
salute you for it sir!