motioneye-project/motioneyeos #1374: Motion Eye File Storage

Appletatoes, 2018-02-10:
Hello,

I can't seem to find any documentation on how to properly setup the Motion Eye File Storage for use with Uploading Media files to another computer on the network.

I am currently trying to use FileZilla with a username and password and a generated certificate I made to ensure an encrypted connection when the Motion Eye on the Raspberry Pi uploads its media files.

My problem is that I keep getting a connection timed out error after about two minutes of waiting to see if the connection works or not by clicking "Test Connection."

Can someone please help me setup a SFTP connection either with FileZilla or through some other means? I would like to keep copies of the media on both the Raspberry Pi's and then also have the media uploaded to a computer on the network of my choosing. What would be the best secure method to do that?

Thank you!
Appletatoes

ccrisan, 2018-02-12:
The SFTP upload method doesn't support private/public keys authentication (as far as I remember). I recommend that you use password authentication for accessing your SSH server.

Appletatoes, 2018-02-12:
Thank you for the quick response!

So it sounds like OpenSSH will be what I'll use. I've read up on it and you can use the public/private key for authentication but sounds like I'll have to use a Windows password and account to login which works for me.

My only other question is, can I still use this method via Upload Media Files and either FTP or SFTP for the method? If so, what is the proper syntax for entering in the Location field? Could you give me an example please?

If I have to do this via Network Share, could you also give me an example of what I should be filling out for the Root Directory field and the Share Name field please?

Thank you very much,
Appletatoes

Appletatoes, 2018-02-21:
I would appreciate any additional input on this. I'm working on a capstone project and am only struggling with this last step of the project.

Thank you very much in advanced,
Appletatoes

Chiny91, 2018-02-21:
Is this uploading files to be done whilst you are present ?   Presumably yes, hence your use of FileZilla, which I imagine can store passwords like most utilities of that sort.     FTP does have the benefit of being simple but its not exactly secure across non-private networks and you might be better using scp, sftp or whatever with FileZilla or maybe FreeFileSync which looks good.

Or are you trying to do this unattended, under a script or cronjob or Windows equivalent.   In which case, something over passwordless ssh will do the trick.   I use rsync on my network storage (Linux, Synology) to grab motionEyeOS images with cron.

There are a vast number of web pages explaining passwordless ssh.    Essentially you need to generate a pair of ssh keys on your Raspberry Pi, something like:
`ssh-keygen -t rsa`
You can view the generated public key with something like:
`cat /root/.ssh/id_rsa.pub`
and then copy it to the appropriate directory of your network box, if non-Windows, maybe:
`/some user path/.ssh/authorized_keys`
I haven't tried this with sftp but it works fine when testing with ssh in a terminal; and for rsync.

Appletatoes, 2018-02-22:
Thank you Chiny91!

I'll try this out. Yes, I'll want it to work unattended so to grab the files automatically. I see you grab the whole image but I could easily do just the directory of the media files right?

Thank you

Chiny91, 2018-02-22:
Copy the whole media directory ?   Yes, that is what I do, more or less.

I copy `/data/output/*` from motionEyeOS to my network storage, but exclude files under 1 kB and files over 250 kB, as probably duff but certainly not useful pictures.

sfgabe, 2018-03-11:
As of this latest update (201802124) I am also experiencing a hang / timeout on FTP setting test and failed upload. Same settings, same server, was working before the update. Anything change that would effect the FTP connection?

Appletatoes, 2018-03-11:
Hi Chiny91,

So I'm playing around with rsync and I'm using PuTTY on my Windows 8.1 machine and am connected to a raspberry pi with Raspbian installed. I'm trying to run rsync to grab files on Raspbian and have the m transfer over to my Windows 8.1 machine. I'm guessing it isn't working because rsync isn't installed on my Windows 8.1 machine?

I know you mentioned you use rsync to sync the files from your MotionEyeOS to your network storage. Could you elaborate on how I would go about using rsync to transfer files from the MotionEyeOS to my Windows machine or even from the MotionEyeOS to a Raspbian OS?

I know you metioned rsync, and I like the idea of it and using cronjob to automate it. Some additional help or examples would be very appreciated!!

Thank you,
Appletatoes

Appletatoes, 2018-03-12:
Update

I tried making private and public keys and I put everything in right place but am still getting prompted for a password. Any ideas on this? Wouldn't mind some additional help please.

Thank you,
Appletatoes

jasaw, 2018-03-12:
@Appletatoes For SSH key login, copy `/data/etc/ssh_host_rsa_key.pub` to your file server. On your server, add the ssh_host_rsa_key.pub to the ~/.ssh/authorized_keys file.
On your MotionEyeOS, run `ssh -i /data/etc/ssh_host_rsa_key -o UserKnownHostsFile=/data/etc/known_hosts username@file_server_hostname` to verify that SSH key login works.

Mraanderson, 2018-05-01:
`scp -r admin@192.168.1.84:/data/output/* /home/pi/cameras/` where 192.168.1.84 is your Motion-Pi and this is run on the 'collecting' PC (Raspberry Pi in my case) by a cron task.

jasaw, 2018-05-02:
Might be a good idea to add "how to backup to a local file server" to [wiki HowTo section](https://github.com/ccrisan/motioneyeos/wiki/Howtos).

@Appletatoes Did you manage to resolve your ssh-key login issue?

Danara5, 2019-05-21:
I solved this with my Raspberry Pi setup by having MotionEye save the files under the pi user's home directory (/home/pi/motioneye/).  The files are initially owned by root, which is a problem since I wanted to pull them off the pi using rsync.  In MotionEye, I used the file storage option to run a command, and had it change the owner of the files to the pi user.  chown -R pi:pi /home/pi/motioneye/   The -R is for recursive, so it processes the entire directory every time it saves.  On my other machine, I pull the files (and remove them), using rsync, every 10 minutes (via cron) with a bash script containing this

rsync --remove-source-files -av pi@192.168.1.27:/home/pi/motioneye/ /home/dan/motioneye/

My Pi has a very small SD card, so I can't store even 1 day of files.  It would be nice if motionEye had an option to retain files for less than a day (like xx hours).

seaniedan, 2020-02-20:
Or better, if the issue is storage space, have an option to delete the oldest files once the storage hits a given percentage of the storage space (e.g. 90%).

I'm having trouble on MotionEye OS uploading media files via SFTP. I can connect from another machine with the same credentials (via the command line or an FTP program). But the Raspberry Pi running MotionEye OS won't let me. 

I can SSH into the pi and run SFTP with write issues: 
[root@meye-xxxxx ~]# sftp user@xxxxx.com 
Could not create directory '/root/.ssh'.
The authenticity of host 'xxxxxxx.com' (x.x.x.x) can't be established.
ECDSA key fingerprint is xxxxxggggg. Continue (Yes/No) (yes)
Failed to add the host to the list of known hosts (/root/.ssh/known_hosts).
user@xxxxxx.xom's password:
Connected to user@xxxxxxx.com.
sftp>

So the problem is I can't 
[root@meye-xxxxx ~]# mkdir /root/.ssh
because I get
mkdir: can't create directory '/root/.ssh': Read-only file system

Any help appreciated!

starbasessd, 2020-02-20:
Try
turn on Debug mode
After reboot, try your process.
Reboot
Turn off debug
reboot
try setting your SFTP

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

seaniedan, 2020-02-20:
I turned on Debug and rebooted. In MotionEyeOS, I still get:
Accessing the Upload Service failed (78, "")!
However, I can now SSH into the pi and access the ftp through SFTP on the command line without write issues! :-D
Sadly, in MotionEyeOS, I still get:
Accessing the Upload Service failed (78, "")!
Even after another reboot, same error.

starbasessd, 2020-02-20:
Does the folder structure already exist on the SFTP service exist?
Do you have read/write access?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

seaniedan, 2020-02-20:
yes and yes. I can SFTP from the machine fine, read and create directories. I just can't do it through MotionEyeOS.

Chiny91, 2020-02-20:
@seaniedan You'll need to read the latest log file (probably huge if you've left debug on).   Looking at your first comment, you say:

> So the problem is I can't
[root@meye-xxxxx ~]# mkdir /root/.ssh
because I get
mkdir: can't create directory '/root/.ssh': Read-only file system

OK, so make the root directory writeable:
https://github.com/ccrisan/motioneyeos/wiki/Tweaks#partition-layout
Which should sort the first error, then try again and if no joy, move on to the new error.

I usually find it easier to initiate all transfers from the non-motionEyeOS end, as motionEyeOS is lean & mean (for good reason).   This may not be feasible in your case.

starbasessd, 2020-02-20:
Same user?
I just had a similar issue with SFTP to an OpenMediaVault setup on a second
PI.
[ccrisan/motioneyeos] Motioneyeos - Openmediavault (#2298)
Finally turned out to be R/W permissions for the user being used for the
motionEyeOS.
There's instructions and a couple of screen shots.
Check it out and see if that helps.
Otherwise, I'll need a bit more info as to what you are using for the SFTP
server.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-20:
Chiny91
One of the benefits of turning on debug is it enables the R/W function to
all the file systems.
Just FYI, as it allowed him to create the .ssh folder. It can be a anvil
when you need a fly swatter, but sometimes faster.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

seaniedan, 2020-02-24:
I solved this using the following method:
On the command line, use
sftp user@ftp.site
enter your password, and once connected you should see
sftp>
enter the command to print the working directory:
pwd
take the output of the pwd command and put that absolute path into motioneyeos.

seaniedan, 2020-02-24:
(Thank you so much for the 'debug' hint, that also helped!!!)

spydeylord, 2020-09-20:
Few days back i installed motioneyeos in raspberry pi 4 and its working perfectly now. I configured the camera settings and enabled upload service via google drive. But i cant view any recordings or photos in google drive and i dont know to see where its stored. And i am a complete beginner in coding and tech stuff so pls someone help me. This is the first time i am using github.

spydeylord, 2020-09-27:
@ccrisan @Chiny91 @jasaw  help me pls.
I have  mentioned in previous comment.

starbasessd, 2020-09-27:
Very simple:
Open Settings in WebGUI

Open File Storage
Set Upload Media Files to "ON"
Set Upload Pictures and/or Upload Movies to "On"
Select Upload Service to "Google Drive"
(create a folder called "motion" in Google drive)
enter "motion" in Location
Set Include Subfolders to "On"
Leave Clean Cloud "Off"
Click Obtain Key, and follow the process and copy key
Paste key in Authorization Key
Click Apply at the top
Click Test Service
Should come back with message at the top saying Accessing the service was
successful.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

seaniedan, 2021-03-20:
Hello all,
this issue could really be called 'SFTP' problems, as that was the original posters problem and mine.

Thanks to the help of posters above, I got sftp working.. to an extent. While some media gets uploaded fine, on many others I get errors:

2021-03-20 07:08:24: [motioneye]    ERROR: sftp: cURL upload failed on sftp://xxxxxx.com:22//home/xxxxx/xxxxxxx.com/zenphoto/albums/motioneye/front/2021-03-20/07-07-05.mp4: 
2021-03-20 07:08:24: [motioneye]    ERROR: failed to upload file "/data/media/sda1/2021-03-20/07-07-05.mp4" with service sftp: (79, '')
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 1103, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 103, in upload_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 992, in upload_data
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 952, in curl_perform_filetransfer
error: (79, '')

..so I have a cron job which hourly rsyncs the media to the remote server. This uses rsync over sssh and private/public keys authentication. This works fine but is always lagging what is happening on the camera, and doesn't really fix the issue outlined above. 

It would be great if the SFTP option allowed for key authentication, the underlying code does allow for this: http://pycurl.io/docs/latest/callbacks.html#ssh-keyfunction

Any furhter insights would be welcome! 

Thank you all for the useful help.

starbasessd, 2021-03-20:
One option could be to run your rsync script from Run A Command, rather
than use Upload Files.
I don't know if your issue may be related to some of the files not
uploading because of rsync and an active upload going on at the same time.
Does the issue happen to files of a certain size or larger, or at the same
time as the rsync, or any other commonality?
What does your space available on the local drive look like for /data?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

seaniedan, 2021-03-24:
As a workaround, I tried running rsync (or scp) as a command after motion event finishes, but the issue there was, the video hasn't finished being made. I can delay upload with a sleep time, but it's all a bit of a guess. 

Good thinking about looking for commonality with 2 uploads going at the same time. I will look into that.  

Space on the local drive is fine (currently a USB stick, but previously had same issue with MicroSD card), and space on FTP destination is fine. Maybe I'll try a different FTP server as a test.

thebream, 2021-06-03:
FWIW. I also cannot get "Test Service" to work with SFTP (it just hangs for a few minutes).

Am running motionEye (**ME**) addon to Home Assistant OS (**HA**). ME runs in a Docker container under HA.

Testing this from HA shell, SFTP does work but first time it prompts "The authenticity of host . . . can't be established", I select "yes" and it adds the host to known_hosts. No prompt when running SFTP again.

I suspect that when I try this from ME, which is running in a separate container, it has no known_hosts file (not that I can create) and will be hanging on the above prompt in the background.

I have worked around by using FTP instead of SFTP, which in local network is OK for me.

Astroamadeus, 2021-06-03:
> FWIW. I also cannot get "Test Service" to work with SFTP (it just hangs for a few minutes).
>
> Am running motionEye (**ME**) addon to Home Assistant OS (**HA**). ME runs in a Docker container under [...]

I have the same issue and came to the same conclusion. It would be great if this issue could be fixed ;)