motioneye-project/motioneyeos #1375: Recording "Fast Network Camera" devices with a central hub

manianac, 2018-02-12:
I have multiple PiZeroW's running each with a CSI interfaced camera.  I can display them fine, but how can I do motion detection or even save them with my central hub?  The hub is an Ubuntu VM running motioneye.  

When I click open the menu on the web interface, I can only rename, with no options showing for saving / motion detection.

ccrisan, 2018-02-12:
How did you add them to the hub? The best setup is setting the Pi Zeros to run as Fast Network Cameras and adding them to the hub as Network Cameras.

manianac, 2018-02-12:
As an MJPEG camera, using the streaming port 8081.  
With "Fast Network Camera" enabled, if I try to add it to the hub it says "Unsupported Network Camera".
If I try to add it as a "remote Motioneye camera", there are no cameras listed in the dropdown, despite seeing it when I view the ZeroWs web ui.

ccrisan, 2018-02-14:
Try the following setup:

 * your (multiple) Pi Zeros set as Fast Network Cameras (will become Simple MJPEG Cameras **locally**)
 * on your hub, add them as Network Cameras, on port 8081

manianac, 2018-02-18:
That did the trick!  Thank you sir, you've created an amazing project

closed 2018-02-18 by manianac

tunix, 2019-08-03:
Hİ @ccrisan ! I've this setup where I have 1 Raspberry Pi Zero W streaming video as Fast Network Camera with settings adjusted like you documented on Tips & Tricks page. As you wrote, enabling fast network camera disables things like recording, motion detection etc. I created an Ubuntu VM on DigitalOcean, installed motionEye on it and successfully added the Raspberry Pi Zero W camera to it. My expectation is to have a continuous recording of the stream on the Ubuntu VM but I couldn't manage to do it.

When I enable video recordings on the VM, no video is recorded but I have a huge collection of still images. Is this the expected behaviour?