motioneye-project/motioneyeos #1379: File storage

tjspragg, 2018-02-15:
Hi! 

I'm really enjoying using this OS - it's really very user friendly! One issue I am having is with File Storage on Dropbox or Google Drive.

No matter how I try to set it up - I seem to receive the same error 'Accessing the upload service failed !'

Do you have any advice on how to go about fixing this? Any advice would be appreciated!

Tomas

ccrisan, 2018-02-16:
Maybe if you let us know how you configured your upload services, we could offer some help. Screenshots are welcomed.

tjspragg, 2018-02-18:
Thanks! Here's the screenshot! 
[image]

ccrisan, 2018-02-20:
And do you actually have a folder named `home` with a subfolder called `CCTV` in your Dropbox?

tjspragg, 2018-02-23:
Yep - very sure. I had the same issue when using Google Drive and Dropbox
(with identical file structures).

ccrisan, 2018-02-24:
@tjspragg let's see some logs then. `motioneye.log` would be of interest. Please enable debugging in Expert Settings first.

tjspragg, 2018-02-24:
Enabled and attached! :) 
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/1754765/motioneye.log)

ccrisan, 2018-02-27:
@tjspragg you have a lot of errors that are consequences of wrong date/time. Please make sure your system can access the internet and has the correct date/time. My personal feeling, judging by the following errors, is that your DNS server, or in general, your network connection is problematic:

    dropbox: failed to obtain credentials: <urlopen error [Errno -3] Temporary failure in name resolution>

or 
    
    1970-01-01 01:07:06: [motioneye]    ERROR: dropbox: failed to obtain credentials: <urlopen error [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:661)>

closed 2018-02-27 by ccrisan