motioneye-project/motioneyeos #1380: A simple question: Is it possible to stream audio and video? (just stream, without recording)

sk777club, 2018-02-16:
I have read all the discussions about "audio" and still have no right answer.

A simple question: is it possible to stream audio and video with motioneyeos, a Raspberry PI, a PiCam and a USB microphone? (just stream, without recording).

If possible, what do I have to do? Video-stream works fine, but without audio. 

Thanks, regards

SK

ccrisan, 2018-02-19:
The answer is simple: no, there's no audio support and there are no plans to add audio support anytime soon. This refers to both live audio and recording.

closed 2018-02-19 by ccrisan

neocogent, 2019-08-15:
It's too bad this isn't in the plans as it would be very nice to be able to use motioneye as video door/gate system instead of having to have a separate system for that. I mean using the front gate camera to answer door/gate and unlock for access. Seems like maybe a small change to streaming to provide another often needed use scenario

jasaw, 2019-08-15:
@neocogent Fast Network Camera mode now supports audio and video RTSP streaming, but audio control is not exposed on the web interface. It is also experimental. I suspect you can't use it as a door intercom because of the high latency. I'm investigating different ways of reducing the latency.

Anyway, if you want to enable audio, see here: https://github.com/ccrisan/motioneyeos/pull/1764

neocogent, 2019-08-15:
@jasaw  Thx. I'll have a look.