motioneye-project/motioneyeos #1381: RTSP Streams instead of MJPEG?

fictionalname, 2018-02-16:
Hello,

I’m still using your excellent software, andI have a question regarding the streaming video.  

I am running MotionEyeOS on a Pi3  and have 2 cameras plugged directly into the device (one onboard and one USB).  At the moment, I use the Streaming URL option to embed the live MJPEG video in a web site.  

I have read that although the MJPEG stream gives the best quality, it is greedy on bandwidth.  Is there any way to get these streams as RTSP (or another compressed format) to reduce the bandwidth across the network.  I was going to set up FFMPEG on my Windows Server to transcode the streams before passing it on, but this seams overkill if I could get this done in the Pi.    

I don’t know if is possible to install FFMPEG on MotionEyeOS or how to set up the transcoding of 2 streams in a single command?

Hope you can help.

Thanks for your time and hard work.

ccrisan, 2018-02-19:
The MJPEG stream is generated by motion. Unless they decide to add a RTSP stream, it's very unlikely that motionEye(OS) will expose a RTSP stream. Transcoding is not something you'll want to do in CPU, especially on single board computers. If however someone comes up with a hardware accelerated ffmpeg-based transcoding option, we could integrate it as an experimental option, just like Fast Network Camera was at the beginning.

closed 2018-02-19 by ccrisan

fictionalname, 2018-02-19:
Hello, 

Thanks very much for your response.  Fingers crossed this is something which will come in a later build.

Keep up the great work!