motioneye-project/motioneyeos #1384: Can't find Expert Settings in motioneyeos

laphlaw, 2018-02-17:
I flashed motioneyeos-raspberrypi3-20170610.img to my raspberry pi 3. I can't find expert settings anywhere.

I'm using this camera:
https://www.amazon.com/gp/product/B0755FGY1F/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

mb39924, 2018-02-17:
Make sure you enable advanced settings first. It's under General settings.

ccrisan, 2018-02-19:
@laphlaw if @mb39924's answer doesn't help you, we need to see a screenshot. If it does, please close the issue.

laphlaw, 2018-02-19:
Yup that fixed it. Thanks!

closed 2018-02-19 by laphlaw

Essaysee, 2019-10-10:
Ccrisan, am I missing something?
I too am having this issue.
I've attached the screenshot (hopefully) to show that there is no toggle for 'advanced options'.
I'm trying to troubleshoot an issue I'm experiencing with the Motioneye OS addon in homeassistant where I can't get python scripts to run.  I thought if I could get the motion notification accordion to show up, I could try running motion notification scripts from the Raspberry Pi 0 W itself to isolate the issue and see if it was an issue with the addon passing scripts over to the Pi W running motioneye OS. 

Best regards and thank you for your work!
[image]

jhussain7262, 2019-10-10:
I have setup motioneye on Pi Zero W with 2 USB cameras. The issue I have is that if I setup both cameras as PC camera then they do not output any view and screen flickers, but if I set one as PC camera and other off, then they work fine. If I use one as PC camera and other as the BCM codec then both stop working. Any idea how to use these two USB generic cameras. They work fine individually but not together.

Thanks.

Essaysee, 2019-10-10:
@jhussain7262 this seems like a separate issue than the original post, but it sounds like you may be running into a Pi zero W hardware limitation.  In attempts to isolate your issue, try installing motioneye os onto a Pi 3 or another machine/device.  The motioneye OS should be able to handle multiple cameras, just change the hardware your OS installation is running on.

jhussain7262, 2019-10-10:
Thanks, Essaysee.  I have Pi 3B+ and I shall try on that. Can I simply swap the sd card from zero to 3B+ or I have to start from scratch to install on 3B+?

Thanks.

Essaysee, 2019-10-10:
I believe you can actually swap the microSD card into the Pi 3.  Be sure you know the new IP address as that likely will change.