motioneye-project/motioneyeos #1392: Rainbow Screen and never Booting

wayne3503, 2018-02-21:
I have a Raspberry Pi 3 and cant seem to get Motion Eye OS to work at all. Here's the issues I've had so far. At this point I have tried two SD cards and even though I've had the same end result, Ive had different issues pop up along the way.

SD Card 1: 
As mentioned in the provided instructions, and others I came across on the Googles, I have used Win32DiskImager to put the images onto the SD card.
Images Used: motioneyeos-raspberrypi3-20180101 and motioneyeos-raspberrypi3-20171008
I have tried numerous times and after each attempt I'd boot my computer into GParted and delete all partitions, reformat, and even reformatted again when booted into windows, put the image onto the SD card and try again. Wash, rinse, repeat.
Each attempt I would let it sit for the requested couple minutes, but it always ended up being longer. Never was it connecting to the network, or did anything for that matter. Any time I connected my HDI to it I'd either get the giant rainbow screen, or nothing at all but blackness.
I also tried Etcher and that's when I discovered that my SD card was corrupt. I eventually was able to take care of that and put an image on it but still got the same result and then decided to move onto SD card 2.

SD Card 2:
I followed much of what I was doing before with SD Card 1 with the exception that I was ONLY using the motioneyeos-raspberrypi3-20180101 image and was still getting the same result which was after each attempt I would let it sit for the requested couple minutes, but it always ended up being longer. Never was it connecting to the network, or did anything for that matter. Any time I connected my HDI to it I'd either get the giant rainbow screen, or nothing at all but blackness.

Am I missing something? Is this a known issue? I have followed the directions to a T and still having this issue? Any ideas how to correct this? Any help would be much appreciated.

jasaw, 2018-02-21:
Did you decompress the downloaded motioneyeos-raspberrypi3-20180101.img.gz file? The file is gz compressed, so you'll need to decompress it to get motioneyeos-raspberrypi3-20180101.img (notice the lack of gz). Flash this decompressed img file to your SD card.

wayne3503, 2018-02-22:
Yes ID. I am an IT guys so I do know about some of these things..lol It was just the img being put onto the SD card, not the compressed file. :)

jasaw, 2018-02-22:
Just to help you speed up your testing process, you don't need to use GParted or any tool to delete partitions on the SD card before you flash the image on. Flashing the image will overwrite all data on the SD card, including partition tables.

To verify that you have successfully flashed the SD card:
- Flash SD card
- Pull SD card out of PC and plug in to PC again.
- If SD card is flashed properly, you should at least see the "Boot" partition from your windows machine, and there should be "config.txt" and "cmdline.txt" files among other files in the "Boot" partition.

If Win32DiskImager supports verifying SD card after flashing, enable it.

wayne3503, 2018-02-23:
As for flashing the SD card rather than going to GParted: When I'm done flashing I dont get just one drive showing up showing all of the partitions and still show that its "16GB" (like the way a normal drive would behave), it shows 2 drives "E: and F:" they are 900 MB and 20 MB respectively. There is another unlabeled 800-ish MB partition showing is Disk Management as well. 

I do see the "config.txt" and "cmdline.txt" (residing on the F:). 

There is a "verify only" button in Win32DiskImager. I will need to reflash to see if it works which I will try right now and post an update. When I used Etcher, that did have a validation process, but with the few errors I got, t hat didn't mention anything about validation, I wasn't sure if it actually worked or not.

wayne3503, 2018-02-23:
After reformatting the SD card, I loaded the image file into Win32DiskImager and hit the "Verification Only" button and got "Verification failed at sector 0" not sure how reliable that is, but that's what happened. I still went ahead and wrote the image to the SD card. When completed it still says "write successful". I tried to do "Verification Only" after the image was on the SD card and this time I got "Verification failed at sector 2048". I put it into the Pi and booted up, still get the big rainbow screen and nothing happens.

jasaw, 2018-02-25:
Can you please try flashing Raspbian on the same SD card, see if you still get rainbow screen?

wayne3503, 2018-02-28:
Raspbian flashed without error and booted up without error...Im stumped as to why motioneyeos is causing issues. This would lead me to believe that the [brand new] SD card is fine, which would then also lead me to believe that the bad sector error(s) are false negatives or just false?

jasaw, 2018-02-28:
That's really odd. I'm not sure what else to try, except maybe try the latest pre-release: https://github.com/ccrisan/motioneyeos/releases/tag/20180224

wayne3503, 2018-02-28:
Well, I'm glad I'm not the only one stumped by this. I will try that and get back to you. Probably tonight even. :)

wayne3503, 2018-02-28:
Ugh. Still the same thing. That sucks. I really need to get this working. My wife and I do foster care, and to be able to see the kids behavior when their in their room would be huge. We have twin 2 year olds and one of them scratches the others eyes when he's sleeping, the bigger one of the 2 tends to be a little more aggressive than the other, but the other will hit back if it comes to it, being able to see whats going on when they try to tear the room apart, etc. Maybe I just need to buy a new pi, good excuse right? :)

jasaw, 2018-02-28:
Maybe try `dd` command from Linux to write the motionEyeOS image to SD card?
Another option is to run MotionEye on Raspbian. https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian

wayne3503, 2018-02-28:
I originally had motion eye installed but I wanted better video quality
with my HD camera and having MotionEyeOS instead and how it operates seemed
like it would be a better option in comparison. I will try the dd command
and see what I can do with that. Thanks again for all of your help on this.

>

ccrisan, 2018-02-28:
@wayne3503 also check this troubleshooting article: https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#system-wont-boot

Most of the "won't boot" issues are caused by either wrong board (are you 100% sure you have a RPi3?) or wrong OS image burning.

Another idea is to make a screencast of how you actually write the image and we'll try and see if there's anything wrong in your process.

wayne3503, 2018-03-06:
I'll be honest, I dont know what the issue was but my persistence paid off. I GOT IT WORKING!!! Thank you guys so much for the support. I'm looking forward to seeing what this can do for me. :)

ccrisan, 2018-03-06:
Alright, then we can close the issue now.

closed 2018-03-06 by ccrisan