motioneye-project/motioneyeos #1395: Adafruit TFT Pi Display installation

scott4205, 2018-02-21:
Trying to add adafruit repo for pi tft display support. It appears that the MotionEyeOS has the relevant directories stripped out. 

Running this command;
`curl -SLs https://apt.adafruit.com/add-pin | bash`

following errors;

```
/etc/apt/sources.list 
/etc/debian_version
no such file or directory

wget
apt-key
command not found
```
 etc 

Is there a workaround solution to enable the display support or allow the repo onto the OS?

mb39924, 2018-02-21:
Install motioneyeos on top of raspbian would be my guess.

ccrisan, 2018-02-23:
https://github.com/ccrisan/motioneyeos/wiki/FAQ#i-cant-seem-to-find-apt-get-or-any-other-package-manager-what-gives

closed 2018-02-23 by ccrisan