motioneye-project/motioneyeos #1396: Changing admin password on the Shell

Kwansui, 2018-02-22:
I have been playing around with the system for only a few days but I can't seem to find anything about changing the password on the Pi. I am using a Pi3.

I have read the discussions about changing the password for the Web UI which I have managed to do successfully. I also did see the discussions about the browser cache being an issue for password changes. However I cannot find a way to change the password for the Pi. The browser cache issue would not apply here.

I tried using the following:
`mount -o remount,rw /
mount -o remount,rw /boot
passwd root`

I got the correct prompts to change the password and also got the "pass for root changed by root"
But when i reboot it still uses admin with a blank password and nothing else.

Please let me know if I have missed something

ccrisan, 2018-02-23:
The root Unix password is **in sync** with the UI password. If for some reason you want to change your root Unix password, you can do that using `passwd` (no r/w remount is needed, you can safely ignore any warnings you'll see). It will however be changed back as soon as you change it in the UI. See more [here](https://github.com/ccrisan/thingos/wiki/User-Accounts).

Kwansui, 2018-02-23:
Thank you the quick reply.

My UI password has been changed with the admin and username changed to "pi". This change has worked for login via the UI but when I login using the "black and white" screen it still uses admin and blank password.

If I try using `passwd` without remounting, it mentions that it is a Read-only file system. I found the remounting command from the FAQs about sudo access.

maglub, 2018-02-23:
In the FAQ, this is described:

* https://github.com/ccrisan/motioneyeos/wiki/FAQ#how-can-i-log-in-on-sshtty-console

I.e, you set a new admin-password in the web-GUI, and the FAQ says that the ssh password will be effective after a reboot. This is not completely correct. The password is effective immediately.

The password change in the GUI changes the files /etc/shadow and /data/etc/shadow for the users "root" and "admin", and also calls the adminpassword hooks in /etc/adminpasswd.d to take care of any dependencies (i.e samba, but in the future also other possible features). It also creates an sha1sum hash of your password and puts it into /data/etc/motion.conf.

The GUI uses the script "/usr/libexec/meyepasswd", which in turn calls /usr/sbin/adminpasswd.

You will need to do the remount, and you will need to change the password for both the root user and the admin user manually, or use the wrapper mentioned above. You will also need to create the password hash and put it into the application config file motion.conf.

* Example when using "passwd":

```
mount -o remount,rw /
#use the same password for root and admin
passwd admin
passwd root
```

* Example when using the wrapper:

```
MEYE_USERNAME=admin MEYE_PASSWORD=password /usr/libexec/meyepasswd 
```

This will only change the password for the CLI (ssh). Not the application. The application password is set in the file /data/etc/motion.conf

* Example to set the password for the web GUI user "admin" to "password"

```
echo -n password | sha1sum
5baa61e4c9b93f3f0682250b6cf8331b7ee68fd8
```

* Edit /data/etc/motion.conf and change the value after "@admin_password" to this

```
# @normal_password 
# @enabled on
# @show_advanced on
# @admin_username admin
# @admin_password 5baa61e4c9b93f3f0682250b6cf8331b7ee68fd8
# @normal_username user
...
```

* Reboot or restart motioneye

```
/etc/init.d/S85motioneye restart
#or
reboot
```

Kwansui, 2018-02-23:
@maglub Thank you for the elaborate explanation. 

Unfortunately I do not have access to the Pi for the next 2 days so I cannot try something new based what you have explained. I will post my results when I can.

ccrisan, 2018-02-24:
@maglub 
> I.e, you set a new admin-password in the web-GUI, and the FAQ says that the ssh password will be effective after a reboot. This is not completely correct. The password is effective immediately.

You're right. That's no longer the case for some time now. I've corrected it. Thanks.

maglub, 2018-02-25:
@Kwansui You are welcome.

Not so much to test, actually. If @ccrisan has documented this somewhere, I suggeste we can close this ticket.