motioneye-project/motioneyeos #1397: Turn Off LED's Page

jethornton, 2018-02-22:
The [Turn Off LED's](https://github.com/ccrisan/motioneyeos/wiki/Turn-Off-LEDs) page fails to mention that the /boot is mounted read only. New Linux users will be very frustrated to try and edit a file that is read only. A link to the FAQ [Why do I get Read-only file system when trying to edit a file?](https://github.com/ccrisan/motioneyeos/wiki/FAQ#why-do-i-get-read-only-file-system-when-trying-to-edit-a-file) should be included on that page. An even better solution would be a GUI switch to turn off the RPi LED(s) like the one for the camera LED. The V2 camera does not have an led that I can see.

jasaw, 2018-06-01:
Thank you for pointing out the lack of information on the wiki page. I have updated it to include command to remount `boot` as read-write.

@ccrisan What do you think of adding ACT and PWR LED control to the web interface?

ccrisan, 2018-06-05:
@jasaw I guess we could add them to the UI as well.