motioneye-project/motioneyeos #1399: Support for Orange Pi H3/H2+ SoC Family

rradar, 2018-02-22:
Hello Everybody,

since the newest release the orange pi one (price tag $10) is supported it should be easily possible to support also these boards with the same SoC (including the alwinner H3 and the little brother H2+) from the same vendor.

I think the most interesting boards are the Orange Pi ZERO, LITE and PC (pricetags are around $8, $11 and $15).

I'm don't know how motioneyeos is builded, but in the best case it should be (once the targets are set up) just one file depend file with the hardware configurations.
The other option would be a general h3 platform build who is able to choice the right board in the runtime (or just one at first boot) and sets up the available hardware (wifi, lan, usb, mic, csi etc.)

Thank's for the nice work to all contributors and specially to the maintainer @ccrisan

ccrisan, 2018-02-24:
Support for Orange Pi One is already there. Orange Pi PC 2 will be supported soon (I've been recently given such a board). As for the others, see https://github.com/ccrisan/motioneyeos/wiki/FAQ#can-motioneyeos-be-ported-to-insert-your-board-here.

rradar, 2018-02-24:
Hi @ccrisan 
thank's for your answer! The Orange Pi PC **2** has already the _next generation_ SoC - the allwinner **H5**. The biggest advantages (over the H3/H2+ familiy) is in my opinion the _possibility_ for integrated gigabit ethernet.

Narly9999, 2018-03-14:
The Gigabit Ethernet and USB 3.0 ya mean! 8-) USB 3.0 makes WiFi dongle 2 gig and 5 gig faster.  600Mbps Mini Wireless Dual Band 2.4/5GHz USB Wifi Adapter LAN Antenna Network Adapter 802.11ac/a/b/g/n

 Or do I have this totally wrong? 8-/

jeff490, 2020-10-15:
It would be really really awesome to try this os on my Orange Pi PC. I see that the link the creator linked to almost 3 years ago now states that there is no planned addition of new boards. Any help would be appreciated.

starbasessd, 2020-10-16:
Did you try either
https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-orangepizero-20200606.img.xz
or
https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-orangepione-20200606.img.xz

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jeff490, 2020-10-16:
I did actually try the image for Orange pi one. After using Balena etcher to burn it my Orange pi PC would not even connect to my network through the wired connection. So with my limited knowledge, not being able to see it with my router, to get an IP, I'm not able to browse to anything or use putty to connect. Therefore I don't even know if it loaded or how far into the startup it got.I would assume that with the Orange pi one that the image was made for, what I tried would have worked.Sent from my Samsung Galaxy smartphone.

https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-orangepizero-20200606.img.xz

https://github.com/ccrisan/motioneyeos/releases/download/20200606/motioneyeos-orangepione-20200606.img.xz

closed 2022-01-19 by starbasessd